OPCFoundation/UA-.NETStandard #1496: Possible to store ComplexTypeSystem in Session

koepalex, 2021-09-02:
**Type of Issue**
- [ ] Bug 
- [X] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

When working with the ComplexTypesSystem,  it is required to create an instance per Session. This instance needs to be cached to keep it alive. Therefor all application needs to maintain a cache from session to ComplexTypeSystem.

**To Reproduce**
n/a

**Expected behavior**

Extend the session API with a property to save the loaded instance of ComplexTypeSystem to avoid additional external lookup. 

**Screenshots**
n/a

**Log output**
n/a

**Code snippets**

```csharp
private readonly Dictionary<Session, ComplexTypeSystem> _complexTypeSystemCache = new();
// when session is loaded
if (!_complexTypeSystemCache.ContainsKey(newSession))
{
    _complexTypeSystemCache.Add(newSession, new ComplexTypeSystem(newSession));
}
await _complexTypeSystemCache[newSession].Load().ConfigureAwait(false);
// when session is closed
_complexTypeSystemCache.Remove(session);
```

**Environment (please complete the following information):**
n/a

**Additional context**
Add any other context about the problem here.

mregen, 2021-09-16:
Hi @koepalex , would it be ok to have just a generic 'bucket' property in Session to keep the reference to the complex type system, to avoid a circular dependency?
e.g. public object UserContext { get; set; }

koepalex, 2021-09-17:
It would be not that obvious and require casting but yeah it would do the job

cristipogacean, 2021-09-17:
The Session class provides the `public object Handle { get; set; }` property that can be used for this purpose.

e.g.:
```csharp
try {
	var complexTypeSystem = new ComplexTypeSystem(session);
	await complexTypeSystem.Load().ConfigureAwait(false);
	session.Handle = complexTypeSystem;
}
catch (Exception ex) {
	_logger.Error(ex, "Failed to load complex type system for session '{id}'", session.SessionId);
}
```

Do you think this is enough or we would need a second UserContext like property?

mregen, 2021-09-20:
@cristipogacean good catch, that should do it. @koepalex what do you think?

mregen, 2021-09-23:
@cristipogacean @koepalex hopefully not in violent agreement, closing. Please reopen if not happy!

closed 2021-09-23 by mregen