OPCFoundation/UA-.NETStandard #1497: Acknowledge Alarm- how  identify the Event ID

boualid, 2021-09-02:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [X] Help wanted

**Describe the Issue**
hello,

Currently i am developing OPCUA client using Node Red. I am able to read the event/Alarm from OPC UA server.
now i want to acknowledge the same , so i had tried create a data flow, with OPCUA IIoT nodes , that allow to call acknowledgement.
please find in the below screen shot the content of the node function:
Remark: i took the value of eventID from UA Expert.

[image]

the problem is when i execute the flow i get the below error message:

[image]

it seems that function acknowledge did not accept the Hexadecimal format.
so could you please  help me to identify the Event ID or how can i convert the Hexadecimal EvenID to Byte String.
thank you

Zigby, 2021-09-02:
According to the dokumentation a different error code Bad_EventIdUnknown should be returned for an invalid eventId. It seams to be the ObjectId that is the problem.

https://reference.opcfoundation.org/Core/docs/Part9/5.7.3/

HTH
Zig by

mregen, 2021-09-07:
Hi @boualid, could you specifiy which client/server you are using? Have you checked the quickstart [Alarm&Events](https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/Workshop/AlarmCondition) server and there is a [OPC PLC](https://github.com/Azure-Samples/iot-edge-opc-plc) server which recently added A&E.

EthanChangAED, 2023-07-07:
Closing this issue. Feel free to file a new one in Discussion-Q&A if there is more question.

closed 2023-07-07 by EthanChangAED