OPCFoundation/UA-.NETStandard #1499: Deadband time limit

MikaelSkov, 2021-09-06:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [X ] Help wanted

**Describe the Issue**

I'm using the OPC UA .net client to connect and monitor items and an OPC UA server. I'm using absolute deadband filters on the various tags and that works as intended. Sometimes I have values that does not change over a long period of time, so I would like to set a time limit on the deadband filter, so that even though a value has not changed it will not be filtered out. I have been though the reference guide, but I don't find anything on this neither on the datachangetrigger nor deadband filter.

**To Reproduce**
4. See error

**Environment (please complete the following information):**
 - Runtime: [e.g. .NET 4.6.2, .NET Core 3.1]
 - Component: [e.g. Opc.Ua.Core]
 - Server: [e.g. Reference Server]

mregen, 2022-01-21:
Hi @MikaelSkov, I have seen people using SetTrigger to get timely updates. Another option might be [`ResendData`](https://reference.opcfoundation.org/v104/Core/docs/Part5/9.2/), which to my current knowledge is not supported by our samples.

EthanChangAED, 2023-07-07:
Closing this issue. Feel free to file a new one in Discussion-Q&A if there is more question.

closed 2023-07-07 by EthanChangAED