OPCFoundation/UA-.NETStandard #1500: Running an OPC UA server in kubernetes

kristianmo, 2021-09-06:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [X] Help wanted

**Describe the Issue**

The certificate check for the SDK will verify the current host and if it does not match the current certificate it will generate a new valid one. 

Running as a standalone service this works fine with the pki directory. 

Running in a docker container locally or in a cloud it works ok by mapping the pki directory to some configured certificates in a volume. The host name may not be stable, but usually the IP can be made stable and put into the certificate.

But kubernetes adds more networking routing. The investigation so far suggest the only available check is the hostname, but the hostname = the pod name and will be different every new deployment/change, inherently unstable.

1) Has anybody had success deploying an OPC UA server in a kubernetes cluster?

2) Is there a configuration that can be used to handle this use case? Without breaking security

**Expected behavior**
It is possible to deploy an opc ua server in kubernetes with a stable certificate, i.e. it is not regenerated every deployment.

**Environment (please complete the following information):**
 - OS: [mcr.microsoft.com/dotnet/aspnet:5.0]
 - Development environment: [Visual Studio 2019 16.7.2]
 - Runtime: [.NET Core 5.0]
 - OPC Package Release Version [latest]
 - Component: [Opc.Ua.Core]
 - Server: [Reference Server]
 - Client: [UA Expert]

ZbynekZ, 2021-09-06:
I think that this is not an implementation or deployment question. It is something that probably isn't covered by existing OPC specifications, so the work should start there - by specifying how it should actually work in such environments as Kubernetes. I suggest you open an issue in OPC Foundation issue tracking, https://mantis.opcfoundation.org/ .

PaulHunkar, 2021-09-06:
As I understand it - when you deploy an instance of a docker container or a Kubernetes, you are deploying a new instance of a server (or client).  In OPC UA every server (or client) needs to have a unique Application Instance Certificate (for security).  So I would think as part of the deployment process you need to generate a new Application Instance Certificate that matches the IP address/domain name etc for the node. If you deploy two instance of them then I expect to have two different Application Instance Certificates.  The other aspect of security could be discussed (is it automatically part of a trusted domain? does each deployment have the same trust lists? .....)  

I think there are a lot of points related this type of deployment that should be discussed in specifications, the issue should be a feature request.

kristianmo, 2021-09-07:
Thank you, orchestrators are a more and more common deployment pattern in IT. I think for a standard that bridges OT/IT it would make sense to support it.

closed 2021-09-07 by kristianmo