OPCFoundation/UA-.NETStandard #1506: A certificate which does not present a revocation list endpoint should not be refused for this reason.

GregoireG-C, 2021-09-10:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [X] Compliance 
- [X] Question 
- [ ] Help wanted

**Context**
Our server fails some tests in Security/Security Certificate Validation compliance unit.

**Issue**
A certificate which does not present a revocation list endpoint should not be refused for this reason. Idem for its CA.

List of scrypts impacted :
- Security | Security Certificate Validation | Test Cases |008.js
- Security | Security Certificate Validation | Test Cases |042.js
- Security | Security Certificate Validation | Test Cases |043.js
- 
**To Reproduce**
Can't be easily reproduce since we use a personal certificate validator

**Expected behavior**
The server should accept the client connexion in these tests and the tests should be OK

**Explanation**
According to the OPCUA specification part 4 Table 106 - Certificate Validation Steps :
```
Find Revocation List Bad_CertificateRevocationUnknown 

Bad_CertificateIssuerRevocationUnknown
AuditCertificateRevokedEventType

Each CA Certificate **may** have a revocation list. This 
check fails if this list is not available (i.e. a network 
interruption prevents the application from accessing the 
list). No error is reported if the Administrator disables 
revocation checks for a CA Certificate
This error may be suppressed
```

This means that if the CA certificate have a revocation list, then we check its availability. But if it has not, there is no reason to refuse it. So on my own, i think that the tests 8, 42 and 43 tests more than the OPCUA specification.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019 16.7.2
 - Runtime: NET 4.6.2
 - OPC Package Release Version 1.4.362.42
 - Server : Our software using the stack
 - Client : CTT Version 1.4.9.396

mregen, 2021-09-16:
Hi @GregoireG-C , this topic has been discussed multiple times in the security working group. The result of these discussions is reflected in the CTT and shall be used as your reference to pass certification.  
However, I was not aware of this new requirement, I believe also our implementation doesn't check for the AIA or CRL distribution point in a CA cert. Neither does the GDS sample produce CA certs which such entries, since it requires to provide a globally accessible Url.
I will double check with the CTT team.

GregoireG-C, 2021-09-16:
Thanks, I look forward for the double check,

Can you assert my hunderstanding (it' very important for us since it will generate a specific développement): 
According to the CTT, if the client certificate is **not** signed by a CA (self-signed certificate), It must present a valid clr endpoint.
If it is signed by a CA then, the certificate may not present an endpoint **but** the signing CA certificate must present a valid clr endpoint.

mregen, 2022-01-14:
Hi @GregoireG-C, I hope you were able to resolve the issues in the meantime. I just ran the tests with most recent code on version 1.4.9.398 and the behavior of tests 
`Security | Security Certificate Validation | Test Cases |042.js`
`Security | Security Certificate Validation | Test Cases |043.js`
is controlled by a server administrator. To allow for No CRL it can be ignored by setting the ValidationOption in the security configuration.
For test 008 it will be addressed with PR #1632