OPCFoundation/UA-.NETStandard #1509: ReadNode BadNodeIdUnknown Exception With Certain Tags(Nodes) In KepserverEx Opc Ua Server

AyberkAltuntabak, 2021-09-14:
**Type of Issue**
- [] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [ ] Help wanted

I am building a middleware to fetch node data from Kepserverex OPC UA server. All fetching process behaviour is normal until I try to fetch self configured tag. There is no problem with getting references of those tags. However, I cant get node attributes of given tag. 

When I try to fetch a node(Actually A Tag) with Session.ReadNode method ,it throws an Exception Called BadNodeIdUnknown. 

I know that node that I am trying to fetch is exist. I can see it on Refence Opc Client Application. 

I suppose that this is not a bug but expected behaviour. There is no attribute in the given node even if It is Variable Type Node. My question would be if it is possible that configuring custom node classes with custom attributes. 

**To Reproduce**
Steps to reproduce the behavior:
1-Install Kepserverex.
2-Add A Channel
3-Add A Device To That Channel (Siemens TCP/IP Ethener Connection - S7-1500)
4-Right Click to Device Check Simulation Mode
5-Open Refence Client.
6-Connect to KepserverEx Opc Server
7-Try To Examine Tag's Node Attributes.

**Screenshots**
[image]
[image]

**Log output**
If applicable, add log output to help explain your problem.

**Code snippets**
If applicable, add a code snippet to help explain your problem.

**Environment (please complete the following information):**
 - OS: [Windows 10]
 - Development environment: [Jetbrains Rider]
 - Runtime: [.NET 5]
 - OPC Package Release Version [e.g. 1.4.363.107]
 - Component: [Opc.Ua.Client]
 - Server: [Kepserverex Opc Ua Server]
 - Client: [Reference Client]

**Additional context**
Add any other context about the problem here.

mregen, 2021-09-21:
Hi @AyberkAltuntabak, are you able to read the tags with another client, e.g. UA Expert? It looks like the license is expired, could this be the reason for failure?

AyberkAltuntabak, 2021-09-23:
@mregen Thanks for the reply. You are right. One of possible reasons why you might not be able to fetch Siemens Tags from Kepserverex OPC Server is License Issue. I have student evaulation license so my license for siemens TCP IP drive is for only like 3 hours . When I stop service and restart , my issue is resolved.

closed 2022-01-21 by mregen