OPCFoundation/UA-.NETStandard #1513: ExtensionObject with TypeId Identifier String

Nick135, 2021-09-16:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

My implementation to read and write complex data types (structs) with Siemens S7-1517 works very well.
Now we switch to the new S7-1518 and TIA-V17.
Read and Write of single basic data types works fine, but for strucs I get a "BadTypeMismatch".

Per Wireshark I found a differ of my write request to the UA-Expert write request.

In my write request there is a "Two byte encoded Numeric" as TypeId Encoding Mask and in the UA-Expert request is the Encoding Mask "String"

**To Reproduce**
Steps to reproduce the behavior:

1. Connect to Server '....'
3. Browse Address space '....'
3. Read Struct (ExtensionObject)
4. Write Struct (ExtensionObject)
4. See error (BadTypeMismatch)

**Screenshots**
[image]

**Code snippets**
```C#
public void WriteStructUdt(String nodeIdString, DynamicStruct myStruct)
{
	//Create a NodeId from the NodeIdString
	NodeId nodeId = new NodeId(nodeIdString);

	DataValue dataValue = null;

	//This is for regular Struct/UDT
	if (!myStruct.IsArray)
	{
		//Create an ExtensionObject from the Structure given to this function
		ExtensionObject writeExtObj = new ExtensionObject();

		//Parse dataToWrite to the byte array
		byte[] bytesToWrite = ParseDataToByteArray(myStruct);

		//Copy data to extension object body
		writeExtObj.Body = bytesToWrite;

		//Turn the created ExtensionObject into a DataValue
		dataValue = new DataValue(writeExtObj);
	}
	//This is for array of Struct/UDT
	else
	{
		int arrDim = myStruct.ChildNodes.Count;

		//Declar array of extension objects
		ExtensionObject[] exObjArr = new ExtensionObject[arrDim];

		//Split nodes containing data to write
		for (int i = 0; i < arrDim; i++)
		{
			//Parse dataToWrite to the byte array
			byte[] bytesToWrite = ParseDataToByteArray(myStruct.ChildNodes[i]);
			
			//Copy data to extension object body
			exObjArr[i] = new ExtensionObject();
			exObjArr[i].Body = (Object)bytesToWrite;
		}

		//Turn the created ExtensionObject into a DataValue
		dataValue = new DataValue(exObjArr);
	}

	//Create a WriteValue
	WriteValue writevalue = new WriteValue();

	//Setup for the WriteValue
	writevalue.NodeId = nodeId;
	writevalue.Value = dataValue;
	writevalue.AttributeId = Attributes.Value;
	//writevalue.TypeId

	//Creat a WriteValueColelction
	WriteValueCollection valuesToWrite = new WriteValueCollection();

	//Add the created value to the collection
	valuesToWrite.Add(writevalue);

	try
	{
		//Create a StatusCodeCollection
		StatusCodeCollection results = new StatusCodeCollection();

		//Create a DiagnosticInfoCollection
		DiagnosticInfoCollection diag = new DiagnosticInfoCollection();

		//if (System.Diagnostics.Debugger.IsAttached)
		//    System.Diagnostics.Debugger.Break();

		mSession.Write(null, valuesToWrite, out results, out diag);

		//Check result codes
		foreach (StatusCode result in results)
		{
			if (result.ToString() != "Good")
			{
				Exception e = new Exception(result.ToString());
				throw e;
			}
		}
	}
	catch (Exception e)
	{
		//Handle Exception here
		throw e;
	}
}
```
**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019
 - Runtime: .NET 4.6.2
 - Component: Opc.Ua.Core
 - Server: S7-1518
 - Client:  Opc.Ua.Client

mregen, 2021-09-16:
Hi @Nick135, just a wild guess, I believe you need to set the type information in the ExtensionObject:

```C#
exObjArr[i] = new ExtensionObject();
exObjArr[i].TypeId = typeId // should be in the extensionobject of the read operation. Maybe you can clone the returned extensionobject
exObjArr[i].Body = (Object)bytesToWrite;
```

Nick135, 2021-09-16:
@mregen Thanks, this was the wright position to set TypeId

closed 2021-09-16 by Nick135