OPCFoundation/UA-.NETStandard #1515: UserTokenPolicy for each server endpoint rather than global setting

kimaluco17, 2021-09-16:
Hello, 

The OPC UA specification allows for each individual endpoint to have its own UserIdentityToken policies, but the `ApplicationConfiguration` object contains a `UserTokenPolicyCollection` that can only be set globally instead of a per endpoint basis.

Are there plans to make each endpoint have a configurable UserTokenPolicy associated with it?

mregen, 2021-09-21:
Sorry, currently there are no plans to support `UserIdentityToken` per endpoint. Could you elaborate on your use case?

jbarbosaneto, 2021-09-21:
Thank you for your response Martin. The main use case is simply giving more granular control of configuration to the user.

For example, the user may want to match the `SecurityPolicyUri` for a particular `UserTokenPolicy` to the same security policy used by each endpoint:
```
<ServerSecurityPolicy>
        <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#None</SecurityPolicyUri>
        # user token policy w/ SecurityPolicyUri set to None here
</ServerSecurityPolicy>
<ServerSecurityPolicy>
        <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15</SecurityPolicyUri>
        # user token policy w/ SecurityPolicyUri set to Basic128Rsa15 here
</ServerSecurityPolicy>
<ServerSecurityPolicy>
        <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#Basic256</SecurityPolicyUri>
        # user token policy w/ SecurityPolicyUri set to Basic256 here
</ServerSecurityPolicy>
...
```

Or the user may only want to use a secure `UserTokenPolicy` if the endpoint is not secured:
```
<ServerSecurityPolicy>
        <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#None</SecurityPolicyUri>
        # user token policy w/ SecurityPolicyUri set to Basic256Sha256 here
</ServerSecurityPolicy>
<ServerSecurityPolicy>
        <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15</SecurityPolicyUri>
        # user token policy w/ SecurityPolicyUri set to None here
</ServerSecurityPolicy>
<ServerSecurityPolicy>
        <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#Basic256</SecurityPolicyUri>
        # user token policy w/ SecurityPolicyUri set to None here
</ServerSecurityPolicy>
...
```

mregen, 2022-01-21:
Hi @jbarbosaneto , at this time I see the only option to override the server method GetUserTokenPolicies and do some custom stuff. The basic implementation already populates the policy None with Basic256 but others default to the policy in use. I keep the issue open as a feature request...

closed 2023-07-07 by EthanChangAED