OPCFoundation/UA-.NETStandard #1521: Certficate signed with intermediate CA cannot be validated

symarchand, 2021-09-22:
The GDS signs the certificate for a server. Thie server cannot then validate the certificate chhain.
In this case, we use an intermediate CA.
After investigation, the server can validate the same certificate signed with openSSL using the same root and interediate CA.

The comparison between certificate signed using openSSL and with the GDS shows a difference in the X509v3 Authority Key Identifier, that consists in 2 or 3 fields depending of the case: 
- KeyId: key of the certificate
- DirName: subject of the CA
- Serial: serial number of the CA

For openSSL: 
- KeyId: empty
- DirName: subject of the **root** CA
- Serial: serial number of the **intermediate**  CA

For GDS:
- KeyId: key of the certificate
- DirName: subject of the **intermediate** CA
- Serial: serial number of the **intermediate**  CA

Looking at the RFC 5280, in section 4.2.1.1 
```
   id-ce-authorityKeyIdentifier OBJECT IDENTIFIER ::=  { id-ce 35 }

   AuthorityKeyIdentifier ::= SEQUENCE {
      keyIdentifier             [0] KeyIdentifier           OPTIONAL,
      authorityCertIssuer       [1] GeneralNames            OPTIONAL,
      authorityCertSerialNumber [2] CertificateSerialNumber OPTIONAL  }

   KeyIdentifier ::= OCTET STRING
```

We already worked on a fix, I will propose a solution later this week. If I underrstand correctly, I should make a pull request to do so. Am I right ?

mregen, 2021-09-22:
Hi @symarchand, 
For GDS:
- Authority Key Id: key of the certificate  --> should be subject key identifier of intermediate CA? is this the issue?

Yes, please provide a PR.
Cheers,
Martin

symarchand, 2021-09-23:
Hi @mregen, 
From what we tested at the moment, it works if the Key Id is empty, the DirName (authorityCertIssuer) is set with the subject of the root CA, and the Serial (authorityCertSerialNumber) is set with the serial number of the intermediate CA.

In the RFC 5280, we can also read : 
```
The identification MAY be based on either the
   key identifier (the subject key identifier in the issuer's
   certificate) or the issuer name and serial number.
```
I am not sure if the "or" is exclusive. If it is the case I understand that Key Id must be empty in our case.

Regards, 
Sylvain

mregen, 2022-01-14:
@symarchand finally I came across the similar issue and found the problem is that the AKI is not filled out correctly.
Our cert code fills in the SubjectName of the issuer in the AKI, but it should be the IssuerName of the issuer.

OpenSSL is more strict on checking the conditions than Windows, so the cert chains we create with the gds library cannot be validated on openssl once it contains a SubCA.

The fix is to fill the AuthorityKeyIdentifier with the IssuerName of the issueing certificate instead of the SubjectName of the issuer. So basically the AKI will contain the Subject of the 'grandparent'. 

A fix will be provided.  Apparently it were also possible to only use keyId and serial in the AKI. Please let me know if the fix works for you, once provided the update.

closed 2022-01-14 by mregen