OPCFoundation/UA-.NETStandard #1532: The server library exposes some nodes with HasTypeDefinition reference to the same node.

mrsuciu, 2021-10-04:
**Type of Issue**
- [x] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

The server library exposes some nodes with HasTypeDefinition reference to the same node.
e.g BaseObjectType\FolderType\FileDirectoryType\CreateDirectory

**To Reproduce**
The issue can be reproduced for any method in the objects from Opc.Ua.Core that has the m_typeDefinitionId = m_nodieId.
In BaseInstanceState.cs - PopulateBrowser method will add the m_typeDefinitionId to the list of refernces regardless of the fact that it is the same with curent node NodeId.
The question is why the methods have the same NodeId as their TypeDefinitionId when they are generated by the model compiler. 

**Expected behavior**
Such circular reference should be avoided since one can enter an endless browse operation 

**Screenshots**
[image]

mrsuciu, 2021-10-13:
This happens because the following values in Opc.Ua.PredefinedNodes are as follows

[image]

mregen, 2021-11-02:
The Opc.Ua.Core still needs new nodeset created with the fix

PR #1574 (2021-11-02): Mc issue 94
fixes #1532 
fixes https://github.com/OPCFoundation/UA-ModelCompiler/issues/94
PR #1576 (2021-11-04): Fix binary export of surplus typedefinition
- HasTypeDefinition and TypeDefinitionId should only exist for Object/Variable in .uanodes
- fix binary export for .uanodes
- #1574 missed the binary output case, only fixed xml export
- fixes #1532 
- fixes https://github.com/OPCFoundation/UA-ModelCompiler/issues/94
- fix the default and sample models in this repo
commit 7acddc09 (2021-11-04): Fix binary export of surplus typedefinition (#1576)

- HasTypeDefinition and TypeDefinitionId should only exist for Object/Variable in .uanodes
- fix binary export for .uanodes
- #1574 missed the binary output case, only fixed xml export
- fixes #1532 
- fixes https://github.com/OPCFoundation/UA-ModelCompiler/issues/94
- fix the default and sample models in this repo

note --> ModelCompiler needs to use the updated core lib to produce the same output
closed 2021-11-04 by mregen