OPCFoundation/UA-.NETStandard #1533: Performance degradation in dynamic node sets

hampld, 2021-10-05:
**Type of Issue**
- [x] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

Implementing OPC 40501-1, specifically a production plan with multiple jobs, every job having multi part sets and many parts in every run at a high throughput speed. Creating the parts dynamically requires a lock. When a job (and all dependent nodes) need to be deleted, locks are needed as well. When at the same time a new job produces new pieces, the requirement for locks for every operation leads to performance degradations to the point of the server crashing.

**To Reproduce**
Steps to reproduce the behavior:
1. Implement OPC 40501-1 with dynamic nodes in job/partset/part
2. Simulate a high throughput scenario (100s of pieces per minute)
3. Rolling delete old jobs
4. Observe performance degradation with any standard client

**Expected behavior**
SDK can add and remove dynamic nodes in a thread safe manner.

**Code snippets**
Code snippets have been provided to @AlinMoldovean

messerlia, 2021-11-11:
The locks are required by design of OPC UA Stack of OPC Foundation because it relies on a SortedDictionary, which is accessed from within a lock, instead of using a thread safe implementation of a dictionary implementation like ConcurrentDictionary which works without any locks.
If these locks are missed, the server crashes with a NullReferenceException in the SortedDictionary.

mregen, 2022-01-21:
At this time there is no sample code for Add/Remove Nodes and references and a specific NodeManager that can be optimized. Keep request as enhancement.

GoetzGoerisch, 2025-12-03:
Thank you for your update.
Does that mean that dynamic address space feature is not planned for .NET of the OPCF and thus the feature will not be usable in products.