OPCFoundation/UA-.NETStandard #1534: How to unit test when using this library?

andyfurniss4, 2021-10-05:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ x ] Help wanted

**Describe the Issue**

I have a class which depends on an OPC UA session from your library. This session is created by a session manager class at runtime. I would like to unit test my class which depends on the Session but I am struggling to find a way to do this with the code structure you've provided in your library. You are not using an interface for the Session class which means I can't mock any of the calls my class is making.

How would you advise that I unit test my code when using your library? Is there some sort of dummy session I can use?

mregen, 2022-01-21:
Hi @andyfurniss4, the ask is to provide an ISession interface as abstraction?
Other option is to use the client/server fixture in the unit tests, it comes with more overhead and its sometimes flaky in CI builds.

KitsKlip, 2022-02-09:
@andyfurniss4 
We have the same problem and have done exactly what @pgrimmert has suggested but it is clunky and does not work well in all scenarios. You end up with weird code that is not always clear to all looking at the intend of the tests.

So are trying to add an contribution for this with this [PR](https://github.com/OPCFoundation/UA-.NETStandard/pull/1702)

Hope this can make it into the library and help others as well

andyfurniss4, 2022-02-09:
@mregen An ISession interface would be good and would help a lot with unit testing.

Nice work @KitsKlip! Hopefully your PR can get approved and merged in.

pgrimmert, 2022-02-09:
Finally, glad to see that there is effort to improve the lib. My suggestion was of course just a workaround in order to write unit tests at all. Since the lib is not well structured. 
Looking forward to see changes to the lib.

mregen, 2022-12-11:
`ISession` was added in 1.4.371.41. Please provide feedback.

closed 2022-12-11 by mregen