OPCFoundation/UA-.NETStandard #1564: Question for forcing

simonf5, 2021-10-29:
**Type of Issue**
- [x] Enhancement 
- [x] Question 

**Describe the Issue**

We have a use case on OPC UA Client to terminate a connection to an OPC UA server after the server certificate is added to the CRL during runtime, due to OPC UA Client is still maintaining valid connections to other OPC UA servers. Currently the OPC UA Stack caches validated certificates.  This means that a revoked certificate is never reverified.  Subsequent connections to an OPC UA server that has had its certificate revoked would continue to connect indefinitely, unless the process is restarted.  We learned to call update security configuration to reinitialize the stack which as a side effect clears the cache, and so we periodically calling it. However, we worry that either the side effect could be removed in future versions, or the reinitialize could disconnect existing connections, either of which would be a showstopper for us.  Furthermore, in an environment with many connections, the cache is required for performance reasons, and purging the entire cache at once is suboptimal.

We would like to know if there is a proper way to purge a certificate out of the cache during runtime.

Thank you!

**Code snippets**
```
   protected virtual async Task InternalValidate(X509Certificate2Collection certificates, ConfiguredEndpoint endpoint)
        {
            X509Certificate2 certificate = certificates[0];

            // check for previously validated certificate.
            X509Certificate2 certificate2 = null;

            if (m_validatedCertificates.TryGetValue(certificate.Thumbprint, out certificate2))
            {
                if (Utils.IsEqual(certificate2.RawData, certificate.RawData))
                {
                    return;
                }
            }
```

**Environment (please complete the following information):**
 - OS:  Windows Server 2019
 - Development environment: Visual Studio 2019
 - Runtime: .NET Core 3.1
 - OPC Package Release Version 1.4.365.10
 - Component: Opc.Ua.Core

mregen, 2021-11-02:
Hi @simonf5 , this is a known issue and currently there is no supported practice to update the cached certificates.
The best way to clear the cache is to call the `CertificateValidator.ResetValidatedCertificates()` method.

For detecting a 'state change' of a cached certificate we need to cache more metadata information (e.g. which errors were suppressed) and maybe a helper function that reevaluates the cached certificates and clears the cache from least recently used certificates. Thoughts?

GregoireG-C, 2022-07-25:
On my own, I have deactivated m_validatedCertificates because it is not compliant with the OPCUA norm. First, we put 10minutes delay that we considered acceptable but the certification team told us to remove it.

mregen, 2022-08-23:
Thanks @GregoireG-C, would it help to add a property to disable `m_validatedCertificates` by default? It is only useful for a UI where you don't want to be asked everytime in a dialog box, but for unattended operation it were better to turn it off by default.

GregoireG-C, 2022-08-23:
Yes, it will help us to add a property. Please do so.

PR #1925 (2022-10-06): Make the use of validated certificates opt-in.
## Proposed changes

Currently, the validated certificates are stored in a dictionary for fast compare of the certificate validation when an application reconnects, until manually cleared by e.g. `ResetValidatedCertificates()`. After discussion it was decided to make this function an opt-in feature to avoid use cases where changes in the trusted store or CRL are not reflected in the list of validated certificates.
The runtime penalty of a full check can be avoided if needed with the 'Opt-In' by configuration or by application setting the property. [... cut]
closed 2022-10-06 by mregen