OPCFoundation/UA-.NETStandard #1601: Alarm Acknowledge

rsoeldner, 2021-11-19:
I do receive alarms and events but struggle to acknowledge them. As an example, the following code should directly acknowledge the alarm. I'm not sure how to receive the `ObjectId` property, which from my understanding is required for acknowledging the alarm.
This issue could be related to https://github.com/OPCFoundation/UA-.NETStandard/issues/1216

For the sake of completeness, the screenshot shows the UaExport fields.
[image]

Would be great to receive some guidance/help.

```
Subscription subscription = new Subscription(m_session.DefaultSubscription);
                subscription.DisplayName = "Console ReferenceClient Subscription";
                subscription.PublishingEnabled = true;
                subscription.PublishingInterval = 1000;

                m_session.AddSubscription(subscription);

                var selectClause = new[]
                     {
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("EventId") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("EventType") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("SourceName") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Time") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Message") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Severity") }, AttributeId = Attributes.Value }
                        };
                var filter = new EventFilter() { SelectClauses = selectClause };
                filter.AddSelectClause(ObjectTypeIds.ConditionType, BrowseNames.BranchId);

               System.Diagnostics.Debug.WriteLine("New Subscription created with SubscriptionId = {0}.", subscription.Id);

                // Create MonitoredItems for data changes
                var list = new List<MonitoredItem>
                {
                        new MonitoredItem(subscription.DefaultItem)
                        {
                            StartNodeId = new NodeId("i=2258"),
                            AttributeId = Attributes.Value,
                            DisplayName = "Server Time",
                            MonitoringMode = MonitoringMode.Reporting,
                            SamplingInterval = 1000,
                            QueueSize = 0,
                            DiscardOldest = true
                        },
                        new MonitoredItem(subscription.DefaultItem)
                        {
                            NodeClass = NodeClass.Object,
                            StartNodeId = new NodeId("ns=4;s=5:Analog alarms"),
                            AttributeId = Attributes.EventNotifier,
                            DisplayName = "Server Alerts",
                            MonitoringMode = MonitoringMode.Reporting,
                            SamplingInterval = -1,
                            QueueSize = 100,
                            CacheQueueSize = 100,
                            Filter=filter
                        }
                };

            list.ForEach(i => i.Notification += OnMonitoredItemNotification);

            Session.AddSubscription(subscription);
            subscription.Create();
            subscription.AddItems(list);
            subscription.ApplyChanges();
            subscription.ConditionRefresh();

                System.Diagnostics.Debug.WriteLine("MonitoredItems created for SubscriptionId = {0}.", subscription.Id);
            }
            catch (Exception ex)
            {
                System.Diagnostics.Debug.WriteLine("Subscribe error: {0}", ex.Message);
            }
        }

        private void OnMonitoredItemNotification(MonitoredItem monitoredItem, MonitoredItemNotificationEventArgs e)
        {
            foreach (var value in monitoredItem.DequeueEvents())
            {
                if(value.EventFields.ElementAt(2) == "Analog alarms.PreColumnPressure_HighAlarm")
                {
                    CallMethodRequestCollection methodsToCall = new CallMethodRequestCollection();

                    CallMethodRequest request = new CallMethodRequest();
                    //request.ObjectId = e.;
                    request.MethodId = MethodIds.AcknowledgeableConditionType_Acknowledge;
                    //request.Handle = null;
                    request.InputArguments.Add(new Variant(value.EventFields.ElementAt(0)));
                    request.InputArguments.Add(new Variant((LocalizedText)"abcdef"));
                    methodsToCall.Add(request);

                    // call the methods.
                    CallMethodResultCollection results = null;
                    DiagnosticInfoCollection diagnosticInfos = null;

                    m_session.Call(
                        null,
                        methodsToCall,
                        out results,
                        out diagnosticInfos);

                    ClientBase.ValidateResponse(results, methodsToCall);
                    ClientBase.ValidateDiagnosticInfos(diagnosticInfos, methodsToCall);

                      }
                System.Diagnostics.Debug.WriteLine("##:    {3}: {2}, {4}, {5}", value.EventFields.Select(v => v.Value ?? "<null>").ToArray());
                
            }

            foreach (var value in monitoredItem.DequeueValues())
            {
                System.Diagnostics.Debug.WriteLine("{0}: {1}, {2}, {3}", monitoredItem.DisplayName, value.Value, value.SourceTimestamp, value.StatusCode);
            }
            
        }
```

awcullen, 2021-11-20:
First, I subscribe to receive the ConditionId of the AcknowledgeableCondition by adding the following to the select clause
```
new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.ConditionType, AttributeId = Attributes.NodeId}
```
Then I set the CallMethodRequest.ObjectId to the ConditionId field received.

I forgot where I learned this.

rsoeldner, 2021-11-23:
@awcullen thank you. I was able to receive the `objectId`  and partially acknowledge the alarm.
Unfortunately, most of the time, I see the following error message in the OPC UA Wrapper (https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/ComIOP). 
The problem is not reproducible as of now, since I was able to acknowledge 8-10 alarms in a row.  

```
11/23/2021 13:16:29.144 E_ACCESSDENIED: IOPCEventServer.AckCondition
 UnauthorizedAccessException 'Access is denied. (Exception from HRESULT: 0x80070005 (E_ACCESSDENIED))'
```

Is there any known problem\ indication how to resolve this?

EthanChangAED, 2023-07-07:
Closing this issue. Feel free to file a new one in Discussion-Q&A if there is more question.

closed 2023-07-07 by EthanChangAED