OPCFoundation/UA-.NETStandard #1605: Exception "A valid supertype identifier is required" when importing a NodeSet xml

andythinkiq, 2021-11-23:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [x ] Question 
- [x ] Help wanted

**Describe the Issue**

I'm following @NichtEuler example at https://github.com/OPCFoundation/UA-.NETStandard/issues/546#issuecomment-428452384 to import NodeSet files found at https://github.com/OPCFoundation/UA-Nodeset to my server for testing purposes.
So far I have tried with https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/Glass/Flat/Opc.Ua.Glass.NodeSet2.xml and https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/CommercialKitchenEquipment/Opc.Ua.CommercialKitchenEquipment.NodeSet2.xml, and ran into this exception with both of them.

For Glass.NodeSet2.xml, the node that causes the exception is:
```
  <UAObjectType NodeId="ns=1;i=1020" BrowseName="1:GlassMachineIdentificationType">
    <DisplayName>GlassMachineIdentificationType</DisplayName>
    <Documentation>https://reference.opcfoundation.org/v104/Glass/Flat/v100/docs/7.1.2</Documentation>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1012</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=6414</Reference>
    </References>
  </UAObjectType>
```

And for Opc.Ua.CommercialKitchenEquipment.NodeSet2.xml:
```
  <UAObjectType NodeId="ns=1;i=1005" BrowseName="1:CommercialKitchenDeviceType" IsAbstract="true">
    <DisplayName>CommercialKitchenDeviceType</DisplayName>
    <Documentation>https://reference.opcfoundation.org/v104/CommercialKitchenEquipment/v100/docs/5.2.2</Documentation>
    <References>
      <Reference ReferenceType="HasComponent">ns=1;i=5001</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1002</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=6015</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=6016</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=5002</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=5004</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=5003</Reference>
    </References>
  </UAObjectType>
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new server as @NichtEuler described, https://github.com/OPCFoundation/UA-.NETStandard/issues/546#issuecomment-428452384 
2. Use the NodeSet mentioned above to import
3. See exception

**Questions**
1. How do I get around this issue?
2. Is it the standard way to import a NodeSet file?
3. If not, please advise the steps. I'm a newbie to OPC UA and still learning about it. I would really appreciate any pointers.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019 16.7.2 and 2022 17.0.1
 - Runtime: .NET Core 3.1 and .NET 6
 - OPC Package Release Version: 1.4.367.42
 - Component: Opc.Ua
 - Server: Standar Server

mrsuciu, 2021-11-23:
The exception you encountered stems from the fact that prior to importing the node sets  you need to import their dependent node sets. You can identify them by looking for  ex. (https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/Glass/Flat/Opc.Ua.Glass.NodeSet2.xml)
```
 <NamespaceUris>
    <Uri>http://opcfoundation.org/UA/Glass/Flat/</Uri>
    <Uri>http://opcfoundation.org/UA/Machinery/</Uri>
    <Uri>http://opcfoundation.org/UA/DI/</Uri>
  </NamespaceUris>
  <Models>
    <Model ModelUri="http://opcfoundation.org/UA/Glass/Flat/" Version="1.0.0" PublicationDate="2022-01-01T00:00:00Z">
      <RequiredModel ModelUri="http://opcfoundation.org/UA/" Version="1.04.7" PublicationDate="2020-07-15T00:00:00Z" />
      <RequiredModel ModelUri="http://opcfoundation.org/UA/DI/" Version="1.02.2" PublicationDate="2020-06-02T00:00:00Z" />
      <RequiredModel ModelUri="http://opcfoundation.org/UA/Machinery/" Version="1.01.0" PublicationDate="2021-02-25T00:00:00Z" />
    </Model>
  </Models>
```

In this case you should import (https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/DI/Opc.Ua.Di.NodeSet2.xml) and 
(https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/Machinery/Opc.Ua.Machinery.NodeSet2.xml) before importing (https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/Glass/Flat/Opc.Ua.Glass.NodeSet2.xml)

andythinkiq, 2021-11-23:
Thank you very much Adrian. It works! I'll close this now.

closed 2021-11-23 by andythinkiq