OPCFoundation/UA-.NETStandard #1607: Deactivate Authentication and Encryption

matze-code, 2021-11-23:
**Type of Issue**

 Question

**Describe the Issue**

Is it possible to deactivate authentication and encryption in the OPC-UA Client? 
If I use Opc.Ua.Client.Session.Create with parameter useSecurity = false, still the client reports an untrusted certificate of the server.

**Environment:**

OS: Windows 10
Development environment: Visual Studio 2019
Runtime: .NET 4.7.2
OPC Package Release Version 1.4.363.49
Server: Siemens PLC
Client: OPC-UA reference implementation

mrsuciu, 2021-11-24:
@matze-code This happens because the client ApplicationConfiguration instance passed as the first parameter to the   Opc.Ua.Client.Session.Create, has the property ApplicationConfiguration.SecurityConfiguration.AutoAcceptUntrustedCertificates set to false.
Although not recommended in production use, you can additionally set the above mentioned parameter's value to true in order to obtain the desired effect.

ThomasNehring, 2021-11-24:
This may be due to a misconfiguration in the server endpoints, which can enforce encryption (there is a hidden setting for a security policy in the endpoint), or due to the fact that you are using credentials despite of using none-none for the security mode and policy. If the client has to sent secrets to the server during the handshake they must be encrypted, which will happen using the server certificates public key. In that case, the server certificate must be trusted.

mregen, 2022-08-23:
please reopen if not answered.

closed 2022-08-23 by mregen