OPCFoundation/UA-.NETStandard #1610: NodeId CompareTo fails in certain cases when applied to the invalid node id with namespace index 0 and id 0.

ThomasNehring, 2021-11-24:
**Type of Issue**
- [x] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

A clear and concise description of what the issue is.

**To Reproduce**
Steps to reproduce the behavior:
Execute new unit test from pull request #1609.

**Expected behavior**
The new unit test must not fail.

**Screenshots**
n.a.

**Log output**
n.a.

**Code snippets**
see new unit test:
        [Test]        
        public void NodeIdComparison()
        {
            NodeId nodeId = new NodeId(0, 0);
            DataValue nodeIdBasedDataValue = new DataValue(nodeId);

            DataValue dataValue = new DataValue(Attributes.NodeClass);
            dataValue.Value = (int) Attributes.NodeClass; // without this cast the second and third asserts evaluate correctly.
            dataValue.StatusCode = nodeIdBasedDataValue.StatusCode;

            bool comparisonResult1b = dataValue.Equals(nodeIdBasedDataValue);
            Assert.IsFalse(comparisonResult1b); // assert succeeds

            bool comparisonResult1a = nodeIdBasedDataValue.Equals(dataValue);
            Assert.IsFalse(comparisonResult1a); // assert fails (symmetry for Equals is broken)

            bool comparisonResult1c = EqualityComparer<DataValue>.Default.Equals(nodeIdBasedDataValue, dataValue);
            Assert.IsFalse(comparisonResult1c); // assert fails

            int comparisonResult2 = nodeId.CompareTo(dataValue);
            Assert.IsFalse(comparisonResult2 == 0); // assert fails - this is the root cause for the previous assertion failures
        }

**Environment (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Development environment: [e.g. Visual Studio 2019 16.11.1]
 - Runtime: [NET 4.8 and earlier (should not matter) ]
 - OPC Package Release Version [e.g. 1.4.367.42]
 - Component: [e.g. Opc.Ua.Core]
 - Server: n.a.
 - Client: n.a.

**Additional context**
n.a.

mregen, 2021-11-24:
Hi @ThomasNehring , is this a regression or was behavior always not quite right?

ThomasNehring, 2021-11-24:
> in certain cases when applied to the invalid node id with namespace index 0
> and id 0. (Issue #1610)
>
>
>
> Hi @ThomasNehring
> <https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.co
> m%2FThomasNehring&data=04%7C01%7Cthomas.nehring%40sap.com%7Cb9e811b30d644931
> 1c2408d9af4e5002%7C42f7676cf455423c82f6dc2d99791af7%7C0%7C0%7C63773357315016
> 8261%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1
> haWwiLCJXVCI6Mn0%3D%7C3000&sdata=wlEYjdooX7AGC036YlNPuUshzswUDCXDPYicAnd8Lzo
> %3D&reserved=0>  , is this a regression or was behavior always not quite
> right?
>
> -
> <https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.co
> m%2FOPCFoundation%2FUA-.NETStandard%2Fissues%2F1610%23issuecomment-977878627
> &data=04%7C01%7Cthomas.nehring%40sap.com%7Cb9e811b30d6449311c2408d9af4e5002%
> 7C42f7676cf455423c82f6dc2d99791af7%7C0%7C0%7C637733573150178218%7CUnknown%7C
> TWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%
> 3D%7C3000&sdata=wXC4fFf249TMol7BS0nRfBcvt8zWGUKFWynDkB9k%2BuU%3D&reserved=0>
> <https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.co
> IVVTACQ&data=04%7C01%7Cthomas.nehring%40sap.com%7Cb9e811b30d6449311c2408d9af
> 4e5002%7C42f7676cf455423c82f6dc2d99791af7%7C0%7C0%7C637733573150178218%7CUnk
> nown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXV
> CI6Mn0%3D%7C3000&sdata=DBVwW66SgsgnkMEupYsqaOtOTLv5mfVm5pSqY6RcuTI%3D&reserv
> ed=0> .
> <https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fapps.appl
> e.com%2Fapp%2Fapple-store%2Fid1477376905%3Fct%3Dnotification-email%26mt%3D8%
> 26pt%3D524675&data=04%7C01%7Cthomas.nehring%40sap.com%7Cb9e811b30d6449311c24
> 08d9af4e5002%7C42f7676cf455423c82f6dc2d99791af7%7C0%7C0%7C637733573150188173
> %7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWw
> iLCJXVCI6Mn0%3D%7C3000&sdata=n8I%2FYWJ4AIAwY3zPBp%2B4e8W61P886HVefJDJRuZhJY0
> %3D&reserved=0>  or Android.

Hi Martin,

 

it seems that the code where this goes wrong has not been changed for at
least six years.

 

BR,

 

Thomas

commit 3dc1d4be (2022-02-18): Update BuiltInTests.cs with special case compare Null NodeId (#1609)

A  new unit test which will currently fail, due to incorrect treatment of node ids with numeric identifier 0 in the namespace with index 0. ( "NodeId nodeId = new NodeId (0,0)." ).

Of course such a node id is invalid. It seems to be present, however, in some PLCs under certain situations. We encountered such node ids in the context of a customer ticket. The node ids were read into instance of type "DataValue" using the read service call. The evaluation of the read result then failed due to comparisons like the ones in the u [... cut]