OPCFoundation/UA-.NETStandard #1616: Delete Node Manager from server/MasterNodeManager

atodkar, 2021-11-28:
**Type of Issue**
- [x ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [x ] Question 
- [ ] Help wanted

**Describe the Issue**

How to delete NodeManager from the MasterNodeManager? I was thinking of overriding MasterNodeManager but then it uses m_NodeManagers private member which maintains this list.

mrsuciu, 2021-12-04:
Just for clarification, do you intent to remove the NodeManagers property from MasterNodeManager class or do you intent to introduce the functionality of removing a certain NodeManager which is present in the NodeManagers property to the MasterNodeManager class?

mregen, 2022-01-21:
I think we need a Add/Remove nodemanger method in the MasterNodeManager. Currently the nodemanagers are fixed after start. It can become quite complicated to remove a nodemanager at runtime, adding a nodemanager might be easier as the code already exists.

mregen, 2022-08-23:
fixed

closed 2022-08-23 by mregen