OPCFoundation/UA-.NETStandard #1625: Own Certificate instead of SelfSigned

vinodpydi, 2021-12-02:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X ] Question 
- [ ] Help wanted

Hi,

Can i use my own certificate and place it in the certificate trusted folder, instead of self signed certificate created by stack

mregen, 2021-12-03:
Hi @vinodpydi , you can use your own certificate, and under the pki folder you need to provide a public key .der file and a .pfx with the private key, the default location is the `pki/own` folder where you find the self signed certs.
You also need to follow certain rules for certs, the subject must match and certain extensions must follow spec, and application uri must be included in extensions.

iconics-janb, 2022-01-18:
Hi @mregen. I think I came across the issue mentioned above. I started getting error "The certificate with subject ... in the configuration is invalid. Please update or delete the certificate from this location: %CommonApplicationData%/ICONICS/pki/own" when I try to connect my client to a server. The last version it was working fine was 1.4.366.38.
I am using self-signed cert and I do have .der file and .pfx file with the private key. So I assume my cert does not meet some new requirements added in 1.4.367.41.
Unfortunately I cannot find what these new requirements are, can you please provide some help? Thank you.

mregen, 2022-01-18:
Hi @iconics-janb, before 1.4.367 the self signed certificate was automatically renewed when it became invalid for some reason. 
This could lead to issues between trusted UA applications or when a CA issued certificate was replaced. Now the cert check on startup requires manual deletion of an invalid certificate to avoid losing trust. Could you check the log file for the certificate validation error?

iconics-janb, 2022-01-18:
Hi @mregen, the log contained this:
1/18/2022 16:41:42.097 Checking domains in certificate. CN=FrameWorXServer, O=ICONICS, DC=JAN-LAPTOP.iconics.cz
1/18/2022 16:41:42.112 The server is configured to use domain 'jan-laptop' which does not appear in the certificate. Use certificate anyway?
1/18/2022 16:41:42.112 ***EXCEPTION*** BadConfigurationError 'The certificate with subject CN=FrameWorXServer, O=ICONICS, DC=JAN-LAPTOP.iconics.cz in the configuration is invalid.

My client's self-signed certificate contained FQDN, while the Quickstart Reference Server I was trying to connect to was using NetBIOS name. Adding NetBIOS name to the domain list in my client's self-signed cert seems to solve the problem, Thank you for explanation.

mregen, 2022-01-19:
We should really provide a better error message here why the cert is invalid.. but at least its in the logs. Starting with 367 applications should have a way to renew the certificate by command line option or UI. The console reference apps got this option recently.

mregen, 2022-04-01:
Assume question answered

closed 2022-04-01 by mregen