OPCFoundation/UA-.NETStandard #1630: Cannot start reference client in Visual studio 2017

lucatassi, 2021-12-09:
**Type of Issue**
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**
Cannot start reference client in Visual studio 2017
These error occurs:
1>------ Inizio compilazione: Progetto: Opc.Ua.Configuration, Configurazione: Debug Any CPU ------
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\Sdks\Microsoft.NET.Sdk\build\Microsoft.NET.TargetFrameworkInference.targets(126,5): error : The current .NET SDK does not support targeting .NET Standard 2.0.  Either target .NET Standard 1.6 or lower, or use a version of the .NET SDK that supports .NET Standard 2.0.
1>Compilazione progetto "Opc.Ua.Configuration.csproj" NON COMPLETATA.
1>ApplicationConfigurationBuilder.cs(794,30,794,48): error CS0103: Il nome 'RuntimeInformation' non esiste nel contesto corrente
1>ApplicationConfigurationBuilder.cs(794,62,794,72): error CS0103: Il nome 'OSPlatform' non esiste nel contesto corrente
1>Compilazione progetto "Opc.Ua.Configuration.csproj" NON COMPLETATA.
2>------ Inizio compilazione: Progetto: Opc.Ua.Client, Configurazione: Debug Any CPU ------
2>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\Sdks\Microsoft.NET.Sdk\build\Microsoft.NET.TargetFrameworkInference.targets(126,5): error : The current .NET SDK does not support targeting .NET Standard 2.0.  Either target .NET Standard 1.6 or lower, or use a version of the .NET SDK that supports .NET Standard 2.0.
2>Compilazione progetto "Opc.Ua.Client.csproj" NON COMPLETATA.
2>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Libraries\Opc.Ua.Configuration\bin\Debug\net462\Opc.Ua.Configuration.dll' non è stato trovato
2>Compilazione progetto "Opc.Ua.Client.csproj" NON COMPLETATA.
3>------ Inizio compilazione: Progetto: Opc.Ua.Client.ComplexTypes, Configurazione: Debug Any CPU ------
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\Sdks\Microsoft.NET.Sdk\build\Microsoft.NET.TargetFrameworkInference.targets(112,5): error : The current .NET SDK does not support targeting .NET Core 2.1.  Either target .NET Core 1.1 or lower, or use a version of the .NET SDK that supports .NET Core 2.1.
3>Compilazione progetto "Opc.Ua.Client.ComplexTypes.csproj" NON COMPLETATA.
3>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Libraries\Opc.Ua.Client\bin\Debug\net462\Opc.Ua.Client.dll' non è stato trovato
3>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Libraries\Opc.Ua.Configuration\bin\Debug\net462\Opc.Ua.Configuration.dll' non è stato trovato
3>Compilazione progetto "Opc.Ua.Client.ComplexTypes.csproj" NON COMPLETATA.
4>------ Inizio compilazione: Progetto: UA Client Controls, Configurazione: Debug Any CPU ------
4>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Libraries\Opc.Ua.Client.ComplexTypes\bin\Debug\net462\Opc.Ua.Client.ComplexTypes.dll' non è stato trovato
4>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Libraries\Opc.Ua.Client\bin\Debug\net462\Opc.Ua.Client.dll' non è stato trovato
4>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Libraries\Opc.Ua.Configuration\bin\Debug\net462\Opc.Ua.Configuration.dll' non è stato trovato
5>------ Inizio compilazione: Progetto: Reference Client, Configurazione: Debug Any CPU ------
5>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Stack\Opc.Ua.Bindings.Https\bin\Debug\net462\Opc.Ua.Bindings.Https.dll' non è stato trovato
5>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Libraries\Opc.Ua.Client\bin\Debug\net462\Opc.Ua.Client.dll' non è stato trovato
5>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Applications\ClientControls.Net4\bin\Debug\Opc.Ua.ClientControls.dll' non è stato trovato
5>CSC : error CS0006: Il file di metadati 'C:\Users\Pak\Source\Repos\UA-.NETStandard\Libraries\Opc.Ua.Configuration\bin\Debug\net462\Opc.Ua.Configuration.dll' non è stato trovato
========== Compilazione: 0 completate, 5 non riuscite, 3 aggiornate, 0 ignorate ==========

**To Reproduce**
Cloned the repository
Followed readme indications:
## How to build and run the reference samples in Visual Studio on Windows

1. Open the UA Reference.sln solution file using Visual Studio 2017.  
2. Choose a project in the Solution Explorer and set it with a right click as `Startup Project`.
3. Hit `F5` to build and execute the sample.

started Reference client project

**Expected behavior**
Reference client starting

**Screenshots**
[https://lh3.googleusercontent.com/pw/AM-JKLXYK0qXyRbnUdgfCuuSPvQA2J78pjNPv4WUWbUmYMnEwYWsGSUGqdKurAm-5jF34jOpTuOJEhAcK3VF5dcKVqRtCKoKnZR2On6to1Yqv3zUInYo53-HXYohd1WgKWYINM06R-EmF0Ku75MEQFPIZu5q3g=w1560-h848-no?authuser=0](https://lh3.googleusercontent.com/pw/AM-JKLXYK0qXyRbnUdgfCuuSPvQA2J78pjNPv4WUWbUmYMnEwYWsGSUGqdKurAm-5jF34jOpTuOJEhAcK3VF5dcKVqRtCKoKnZR2On6to1Yqv3zUInYo53-HXYohd1WgKWYINM06R-EmF0Ku75MEQFPIZu5q3g=w1560-h848-no?authuser=0)

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Microsoft Visual Studio Community 2017 Versione 15.9.41VisualStudio.15.Release/15.9.41+28307.1745
 - Runtime: NET 4.6.2
 - Client: Reference Client

**Additional context**
Sincerly i don't know where to start inspecting the problems

mregen, 2021-12-09:
Hi @lucatassi, I believe you need to install a .NET Core 2.1 SDK: `The current .NET SDK does not support targeting .NET Standard 2.0.`

However .NET Core 2.1 is already end of life and support will be removed. I suggest to try with VS2019 or VS 2022.

lucatassi, 2021-12-09:
I've already installed .net core 2.1 from here... [https://dotnet.microsoft.com/en-us/download/dotnet/2.1](https://dotnet.microsoft.com/en-us/download/dotnet/2.1)

Could i try with 3.1 even with vs2017?

PR #1641 (2021-12-17): Fix VS2017 build issue with Roslynator
- vs2017 now only supports .NET 4.6.2 builds, .NET Core 2.1 is end of life. 
- Roslynator was not properly excluded in VS2017, change the condition to <Choose>
- fixes #1630 
- fix a few lgtm warnings
- fix a few license headers (everything outside of \Stack is MIT)
- fix flaky Hiresclock test.
commit 7ae5c117 (2021-12-17): Fix VS2017 build, lgtm warnings, license headers, flaky CI (#1641)

- vs2017 now only supports .NET 4.6.2 builds, .NET Core 2.1 is end of life. 
- Roslynator was not properly excluded in VS2017, change the condition to <Choose>
- fixes #1630 
- fix a few lgtm warnings
- fix a few license headers (everything outside of \Stack is MIT)
- fix flaky Hiresclock test.
closed 2021-12-17 by mregen