OPCFoundation/UA-.NETStandard #1643: Server cannot provide localization of texts returned from browse operations

elliot-gawthrop, 2021-12-17:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [X] Help wanted

**Describe the Issue**

When a client issues a browse request for a node, the server responds with a list of child nodes that contain `DisplayName` properties set to `LocalizedText` objects. It is not clear how the server can translate these text strings for the requesting clients preferred locale.

**Expected behavior**
I expected that when setting the `OnReadDisplayName` field of nodes where `DisplayName` is set, that this delegate would be invoked during browse operations and allow the server to provide localized text for the clients preferred locale. This delegate is invoked when a client requests attributes for a node, but it is not invoked when a client browses a node.

When that didn't work, I expected that if I override `StandardServer.CreateResourceManager`, and then returned a specialization of `ResourceManager` that overrode both `Translate` functions, these functions would be invoked during the request. However these functions not invoked during browse requests - they only seem to be invoked when the server emits events.

**Environment (please complete the following information):**
 - Runtime: [.NET Standard 2.0]
 - OPC Package Release Version [1.4.367.42]
 - Component: [OPCFoundation.NetStandard.Opc.Ua.Server]
 - Server: [StandardServer]
 - Client: [UA Expert]

EthanChangAED, 2023-07-21:
Hello @elliot-gawthrop, have you get this issue resolved?

elliot-gawthrop, 2023-07-21:
@EthanChangAED as far as I am aware this issue is not fixed. I'm no longer working for the client that I raised this on behalf of.

EthanChangAED, 2023-07-24:
Thank you for the information, @elliot-gawthrop.

closed 2025-12-03 by romanett