OPCFoundation/UA-.NETStandard #1654: Session browse only 100 node

Tombost, 2022-01-04:
Hello,

Why when I use session.Browse the output to ReferenceDescriptionCollection is limit at 100 value ?

I have a PLC with a lot of data and I need to browse the entire node tree to collect all the data, but it only take 100 values

`
ReferenceDescriptionCollection refs = new ReferenceDescriptionCollection();`

`session.Browse(null, null, DB_adr, 0u, BrowseDirection.Forward, ReferenceTypeIds.HierarchicalReferences, true, (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, out cp, out refs);`

the **refs** is limited to 100 values

mrsuciu, 2022-01-05:
@Tombost  This can be a limit of the server running on the PLC applied per Browse service call. You should be able to browse all the address space by calling the BrowseNext service if the server on the PLC returns continuations points besides the references. See the OPC.Ua.Client.Session.FetchReferences(NodeId nodeId) implementation as an example.

mregen, 2022-08-23:
@Tombost browse sample with Browse or NodeCache will be provided in the client console app with #1915