OPCFoundation/UA-.NETStandard #1662: Cannot run CCT 1.03.340.380 against ReferenceServer

elliot-gawthrop, 2022-01-13:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [ ] Help wanted

**Describe the Issue**
The [README](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/ReferenceServer/README.md) for the ReferenceServer states that the [sample CCT configuration file](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/UAReferenceServer.ctt.xml) targets UACTT Version 1.03.340.380, but this version of the CCT is not able to run the configuration file.

The latest version of the CCT is able to run the configuration file, but please could you clarify exactly which version you are using?

Is the any other information on this README out of date too? For example I see that **Standard UA Server** profile is now deprecated, is that still the profile being tested?

**To Reproduce**
1. Run the ReferenceServer
2. Install UACTT Version 1.03.340.380
3. Create a new server project
4. Replace the contents of the .cct.xml configuration file with [UAReferenceServer.ctt.xml](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/UAReferenceServer.ctt.xml)
5. Select the **Standard UA Server** profile, the **Method Server Facet** and the **DataAccess Server Facet**.

**Expected behavior**
I expected the suite to pass without errors.

**Log output**
Many errors, for example:
```
Could not read setting "/Server Test/NodeIds/References/Has References With Different Parent Types"
Setting '/Server Test/NodeIds/References/Has References With Different Parent Types' is not configured.
```

**Environment:**
 - OS: [Windows 10]
 - Runtime: [.NET 4.6.2]
 - CCT: [1.03.340.380]
 - Server: [Reference Server]

mregen, 2022-01-18:
@elliot-gawthrop the README is not up to date. A matching CTT for the configuration is e.g. 1.4.9.398. We are currently working on fixing a few remaining issues, as CTT is always a moving target (new or improved tests) and sometimes has also open issues, there is never guarantee al tests pass. Hope this helps.

mregen, 2022-01-31:
Hi @elliot-gawthrop , the CTT config file has been updated for 1.4.9.398 in #1689. Doc still needs updating...

elliot-gawthrop, 2022-01-31:
@mregen thanks!

mregen, 2022-02-18:
please reopen if necessary...

closed 2022-02-18 by mregen