OPCFoundation/UA-.NETStandard #1678: How to get the DataTypeDefinition of types with optional fields

mwieczorkiewicz, 2022-01-22:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [X] Help wanted

**Describe the Issue**

I know I am probably missing something really simple here, so excuse me if the question is trivial. 

I am trying to develop a simple UA mapper for my application - essentially the end goal is to programmatically create a map between OPC-UA Types and DataTypes between internal types of my application (which are not subject to change as it's a legacy application). 

So, to simplify the process here - I preload predefined nodes (using BinaryResources, but it's the same for XML NodeSets), load the actual NodeSet, iterate through the types (both simple and structural types) in order to recreate those structures in my application and for the most part it works as intended.

Now, for some Complex Types (i.e. custom types from my own NodeSet or some standard structured types like EUInformation ("i=887")) it works exactly as intended - I get my DataType, check if it's a structural one, if it is - I iterate through fields and recreate the structure in my application. 

However, for some DataTypes the DataTypeDefiniton field is null and I can't find a way of populating it (even though looking through the schema files shows that the structure has some fields. The simple example in the standard NodeSet would be LocalizedText - its DataTypeDefinition is always null (even though it's supposed to have two fields - Locale and Text according to the Opc.Ua.Types.xsd and https://reference.opcfoundation.org/v104/Core/docs/Part3/8.5/).

I think that applies to the types with optional fields, although I might be wrong (I am not that experienced with OPC-UA after all). 

**To Reproduce**
Steps to reproduce the behavior:
1. Load predefined nodes
2. Check DataTypeDefinition for EUInformation DataType and LocalizedText
3. EUInformation contains DataTypeDefinition while the LocalizedText complex type has null DataTypeDefinition

**Expected behavior**
All types should reflect the exact structure. Alternatively - what is the standard way of getting the complete structure of standard and custom types loaded dynamically? 

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: VS2019
 - Runtime: .NET 4.6.2
 - OPC Package Release Version 1.4.367.75

mregen, 2022-01-23:
Hi @mwieczorkiewicz , maybe the disconnect here is that built in types are treated special and type information is omitted. LocalizedText is a built in type.

mwieczorkiewicz, 2022-01-23:
Hi @mregen, 

Thanks, that makes sense. However, is there any elegant way to get that structure? Technically I could just check if the types are BuiltInTypes and just use the actual class (from Types.BuiltIn) and then just use this, but perhaps there's a better way?

EthanChangAED, 2023-07-07:
Closing this issue. Feel free to file a new one in Discussion-Q&A if there is more question.

closed 2023-07-07 by EthanChangAED