OPCFoundation/UA-.NETStandard #1688: DeleteApplicationInstanceCertificate() is private method.  It cannot be accessed by USServer.cs.

sinopac, 2022-01-29:
**Type of Issue**
- [x] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**
DeleteApplicationInstanceCertificate() is private method.  It cannot be accessed by USServer.cs.

A clear and concise description of what the issue is.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Sample '...'
2. Connect to Server '....'
3. Browse Address space '....'
3. Subscribe to Node '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log output**
If applicable, add log output to help explain your problem.

**Code snippets**
If applicable, add a code snippet to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. Windows 10, Ubuntu, macOS]
 - Development environment: [e.g. Visual Studio 2019 16.7.2]
 - Runtime: [e.g. .NET 4.6.2, .NET Core 3.1]
 - OPC Package Release Version [e.g. 1.4.363.107]
 - Component: [e.g. Opc.Ua.Core]
 - Server: [e.g. Reference Server]
 - Client: [e.g. UA Expert]

**Additional context**
Add any other context about the problem here.

sinopac, 2022-01-29:
if (renewCertificate)
                {
                    await m_application.DeleteApplicationInstanceCertificate().ConfigureAwait(false);
                }

mregen, 2022-01-29:
Hi @sinopac , this function has a public version starting with the .368 version, which is not yet released. 
I see that in 367 without this API there is no way to force renewal of an expired certificate.

mregen, 2022-02-18:
fixed in latest Nuget