OPCFoundation/UA-.NETStandard #1692: When will 1.4.368 be released and will my contributions be part of this?

htbmw, 2022-01-31:
**Type of Issue**
- [ * ] Question 

I am a little confused about the release process:
1.4.367 was officially released in December 2021 with a tag / label of "release"
I have been expecting the next release to be version 1.4.368 approximately around end of January 2022 as described here:
https://github.com/OPCFoundation/UA-.NETStandard/milestone/7

I am waiting for this release because it should contain contributions I made in this PR:
https://github.com/OPCFoundation/UA-.NETStandard/pull/1656

The PR was assigned to the 1.4.368 milestone indicating that it will be in the next release.
Today I discovered that a prerelease of 1.4.367.95 was released 4 days ago.
I checked it to verify if my contributions are part of it, but it doesn't seem so (as far as I can see from the release notes / history).

Was the 1.4.367.95 prerelease done by accident from an old branch? 
I would have expected a prerelease to be a prerelease for an upcoming release version like 1.4.368.xx.

I need to plan releases for my own projects that reference your packages, therefor the need to verify when my contributions will be released.

mrsuciu, 2022-01-31:
@htbmw There will be soon a preview release of the 1.4.368 version which will be branched from the master branch.  PR#1656 has been merged into the main branch and will be part of the 1.4.368 version. The version 1.4.367.95 is a service release for the 1.4.367 release which contains mainly bug fixes and backports of features from the master branch.

htbmw, 2022-01-31:
Ah ok, thanks for clarifying this!

closed 2022-01-31 by htbmw