OPCFoundation/UA-.NETStandard #1708: DateTime write changed of behaviour

GregoireG-C, 2022-02-11:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [ ] Help wanted

**Describe the Issue**
I missed a change that almost put a regression in the produt I work on which use the opcua implementation for the client and server. 
With the merge V1.03.442 ->V1.4.367.41, the write beahviour of DateTime property changed:

The` public void WriteDateTime(string fieldName, DateTime value)` method of `BinaryEncoder` class has been changed:

`(value.Kind == DateTimeKind.Local){}` has been deleted and then `ToUniversalTime` is always called. This method change the date to UTC if its kind is not UTC (Local ... or undefined).
In our case, the OPCUA client does not have the choice about the date date since our client is coded in C++ (we use a VT_Variant), so the kind is undefined. Indeed it does not make sense for us to specify the kind since it's not known in the OPCUA protocol (the DateTime is an int64 in the OPCUA protocol).
We made the modification to set by default the kind to utc to bypass the transformation in UTC. We consider that it's the user role to know what kind of date it is and we must not change it.

We do not moan about the change of behaviour since it's your right to interpret dateTime as you want but is there anywhere you have notified this modification of behaviour? It took me a little time  to find why my OPCUA client loose one hour each time it wrights a dateTime value and i wonder if there is something as revision list which do the fonctionnality modification inventory made in new version somewhere.

Best regards

mregen, 2022-02-15:
Hi @GregoireG-C, your analysis is fully correct. The change happened when encoding/decoding tests were established. I believe during the history of .NET the behavior and defaults of the DateTime kind have changed, but I don't know exactly. 
Anyway we ended up in a situation where the kind is undefined or Local, but OPC UA always uses Utc and as you mention, the protocol has no information of a 'kind'. The transformation forth and back was not deterministic so the solution was to enforce this 'new' behavior to ensure encoded -> decoded data is comparable and ends up with the same values. This becomes specifically difficult when the boundary values as defined in the spec are used. So the outcome was this change. The burden is now on the app to always create DateTime values with a kind. Hope this helps..

mregen, 2022-04-20:
Hopefully answered, please reopen otherwise

closed 2022-04-20 by mregen

GregoireG-C, 2022-04-21:
Yes, I forgot to answer and close the subject. I thank you for your answer.