OPCFoundation/UA-.NETStandard #1709: Get the namespaceIndex by browse path

xky0007, 2022-02-15:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [x] Help wanted

**Describe the Issue**

I have a PLC which OPC-UA server is enabled and I write my own OPC-UA client to connect it. If I don't know the namespaceIndex of the node I want to read but I know the complete browse path such as "Root/Objects/PLC/Status/Running", how can I read the running node without namesapceIndex? How do I convert the browse path to a node ID and get the correct namespaceIndex?

Thanks.

mrsuciu, 2022-02-15:
@xky0007  You should use the TranslateBrowsePathToNodeIds service call which does exactly what you desire to accomplish.

xky0007, 2022-02-16:
> @xky0007 You should use the TranslateBrowsePathToNodeIds service call which does exactly what you desire to accomplish.

I already tested it, but the result is BadNoMatch. Please check my code.

```

            Byte[] continuationPoint;

            var session = client.Session;

            var startNodeId = new NodeId(Objects.ObjectsFolder);
            var browsePaths = new BrowsePathCollection
            {
                new BrowsePath
                {
                    RelativePath = RelativePath.Parse("/PLC/Status/Running", session.TypeTree),
                    StartingNode = startNodeId
                }
            };
            var responseHeader = session.TranslateBrowsePathsToNodeIds(
                null,
                browsePaths,
                out var results,
                out var diagnosticInfos);

            NodeId id = new NodeId((NodeId)results.First().Targets.First().TargetId);
```

Thanks.

mrsuciu, 2022-02-16:
@xky0007 The code looks good. The only problem that I can think of, is that the browse path  that you pass might not be correct in the BNF form: `"/PLC/Status/Running"`. Aren't the PLC, Status and Running nodes browse name namespace indexes different than zero ? The browse path should be specified in the BNF form by including the namespace index of the node's browse names. Ex: `"3:PLC/3:Status/3:Running" ` if their Browse Names's NamespaceIndexes are 3 for all three nodes.

xky0007, 2022-03-02:
> @xky0007 The code looks good. The only problem that I can think of, is that the browse path that you pass might not be correct in the BNF form: `"/PLC/Status/Running"`. Aren't the PLC, Status and [...]

Thanks @mrsuciu. This went back to the initial question... I don't know the namespace index for these nodes. That's why I need to translate the browse path to get the target node, then I can check the relative namespace index.

mrsuciu, 2022-03-04:
@xky0007 You are correct. My bad.
Can't the namespaceIndex be deduced from the Server->NamespaceArray index value corresponding to the namespace URI?

EthanChangAED, 2023-07-21:
Close this issue. Feel free reactivate if there is more to be discussed.

closed 2023-07-21 by EthanChangAED