OPCFoundation/UA-.NETStandard #1711: StandardServer.RegisterWithDiscoveryServer registers unsafe certificate validator

Antricion, 2022-02-16:
**Type of Issue**
- [x] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

In the Opc.Ua.Server.StandardServer.RegisterWithDiscoveryServer(), the RegistrationValidator_CertificateValidation is added as certificate validator. This is a very, very baseline validation that only checks if the name of the requester and the name contained in the certificate match. Due to the way events work, this overrides any previously registered validation check, thus leading to a very unsafe situation, where any client with expired or untrusted certificates can connect to the server once it registered with the discovery server.

**To Reproduce**
Steps to reproduce the behavior:
1. Create server with own certificate check
2. Server registers with DiscoveryServer
3. Connect from client with invalid (e.g. expired or untrusted) certificate
4. Even if your own certificate check would deny this connection, the check passes as "allowed"

**Expected behavior**
If an own certificate validation is implemented, registering to a DiscoveryServer should not break that. While such a check could work as fallback if no check is present otherwise, it must not override the secure certificate check. As all relevant functions are non-virtual,  at least an option to disable the unsafe check should be available.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019 16.11.19
 - Runtime: .NET 4.8
 - OPC Package Release Version 1.4.365.48, 1.4.367.95
 - Component: Opc.Ua.Server
 - Server: Any
 - Client: Any

**Additional context**

mregen, 2022-02-17:
Hi @Antricion, I tried to reproduce your finding with the Reference servers, but when the server registration starts it loads a seperate instance of the configuration from file. Thus adding the validator for the registration uses a different validator instance, so the registration validator doesn't have a conflict with the original one and issue does not show.

However, in the case that the 'no file' configuration builder was used for the server or if the 'SourceFilePath' for the application config is empty, I see there is the described issue because the existing application configuration is used directly and the additional validator is added to the existing. In such a case rather a 'clone' of the ApplicationConfiguration with a dedicated certificate validator should be used.  
Also I see a unrelated issue for the server registration when a password protected private key is used, it cannot be loaded with the current code. 
Investigating a fix, thanks for the report!

closed 2022-02-18 by mregen

Antricion, 2022-02-18:
Thanks for the fast fix, seems to work fine.

PR #1713 (2022-02-18): Server fix for registration cert validation issue
-always use a clone of the configuration and replace cert validator when registration is active
-fixes #1711
commit efd04655 (2022-02-18): Bug fix for server registration issue (#1713)

-always use a clone of the configuration and replace cert validator when registration is active.
-fixes #1711