OPCFoundation/UA-.NETStandard #1715: Load complex type system faster

PvanIeperen, 2022-02-24:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [ ] Help wanted

**Describe the Issue**

to use complex types from a Opcua server I have  made a client inn C#.
Now I want to load the complex type system with the following syntax,
```C#
 this.m_logger.LogInformation($"\nRetrieving complextype information from the server. One moment please\n\n");
 Stopwatch sw = new Stopwatch();
  sw.Start();
  var complexTypeSystem = new ComplexTypeSystem(this.Session);
  await complexTypeSystem.Load().ConfigureAwait(false);
  sw.Stop();
  this.m_logger.LogInformation($"Pre loading of occmplex type took {sw.Elapsed} seconds");
```
This can take up to 8 minutes.
Is there a way that I can load only the complex types from the server that I need to speed up the loading time?

The final application will run on a iotedge device running on Ubuntu 18 with .Net core 3.1
The testing is done on my laptop that is running on Windows 10.

OPCUA package 1.4.367.95
OPCUA complex type system package 1.4.367.95

mregen, 2022-02-25:
Hi @PvanIeperen, the complex type loader uses node cache which can be quite slow. 8 minutes is really a long time, is it a very slow connection or does the server have so many datatypes? If you can provide more info, there is room for optimization.
One option is to reduce the number of types to load is to load only types as needed. Load has a few signatures which allow to specify the types, or to load only enums.

PvanIeperen, 2022-02-25:
Hi @mregen 
I managed it to read only the structure that I need. dis was done in 3 seconds.
```C#
 spinner.Start();
  this.m_logger.LogInformation($"\nRetrieving complextype information from the server. One moment please\n\n");
  Stopwatch sw = new Stopwatch();
  sw.Start();
  this.complexTypeSystem = new ComplexTypeSystem(this.Session);
  this.complexTypeSystem.LoadType("ns=8;i=100230", true, true).Wait();
  
  sw.Stop();
  spinner.Stop();
  this.m_logger.LogInformation($"Pre loading of occmplex type took {sw.Elapsed} seconds");
```
To find the indexnumber of this structure I had to use UA expert to figure out what it was. 
[image]

When I do a read on the nodeId that I know, without loading the complex type there will be an index-number in typeId field that is 1 higher than the 100230 that I found in the UAexpert.
[image]

Is there a way that I can find out what the index number is before I load the complex type?

mregen, 2022-02-25:
Hi @PvanIeperen , every type has a type id, binary encoding id and xml encoding id (if supported). What you see here is probably the binary encoding id. The complex type loader walks the references and adds the encodings to the encodeable factory. 
In this case you had to read they type id for the node, load the type etc. Better were if LoadType knew how to handle it.
Was the 'LoadType' not able to resolve the type from the encoding id? If not, it should be added as a enhancement.

PvanIeperen, 2022-02-28:
Good morning @mregen,

### Indeed the system cannot resolve it.  if I do:
```C#
  this.complexTypeSystem = new ComplexTypeSystem(this.Session);
  this.complexTypeSystem.LoadType("ns=8;i=100230", true, true).Wait();
```
Every thing works fine.

### But it will be more nice when I can do like 
```C#
this.complexTypeSystem = new ComplexTypeSystem(this.Session);
 this.complexTypeSystem.LoadType("ns=8;s=::Main:ComplexStructVector", true, true).Wait();
```
The NodeId we get from the PLC department. It will be nice if we can use it directly or that we can find correct encoding id from the nodeId.

PR #1915 (2022-10-04): Complex types improvements
## Proposed changes

Improvements for complex types.
- Decode structures with multidimensional arrays
- Perf improvements for type loader using new NodeCache fetch multiple interfaces
- DataTypeDictionary validation when dependencies are used.
- interface for complex type resolver
- improvements of console client to demonstrate browse/read of custom values

## Related Issues

- Fixes #1874 
- fixes #1826 
- fixes #1715 
- fixes #920
- fixes #1848 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also f [... cut]
closed 2022-10-04 by mregen