OPCFoundation/UA-.NETStandard #1724: How to read aliases

andythinkiq, 2022-03-01:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [x ] Question 
- [x ] Help wanted

**Describe the Issue**
I'm browsing nodes on a server that was set up using a NodeSet2 file. The file has an Aliases section that looks like this:
```
  <Aliases>
    <Alias Alias="BOOL">i=1</Alias>
    <Alias Alias="INT">i=4</Alias>
    <Alias Alias="UDINT">i=7</Alias>
    <Alias Alias="REAL">i=10</Alias>
    <Alias Alias="HasTypeDefinition">i=40</Alias>
    <Alias Alias="HasComponent">i=47</Alias>
    <Alias Alias="HasModellingRule">i=37</Alias>
    <Alias Alias="HasSubtype">i=45</Alias>
    <Alias Alias="Organizes">i=35</Alias>
    <Alias Alias="HasProperty">i=46</Alias>
    <Alias Alias="HasEncoding">i=38</Alias>
  </Aliases>
```

How do I programmatically find these aliases in general, and especially the data type ones (BOOL, INT, UDINT, REAL)? I used Matrikon OPC UA Explorer to browse and saw the FindAlias method, but it didn't work and returned "BadNotImplemented: Request operation is not implemented" error.

Any help is greatly appreciated.

**To Reproduce**
N/A

**Expected behavior**
N/A

**Screenshots**
N/A

**Log output**
N/A

**Code snippets**
N/A

**Environment (please complete the following information):**
 - OS: [Windows 10, Ubuntu 20.04]
 - Development environment: [Visual Studio 2019 16.11.10]
 - Runtime: [.NET Core 3.1]
 - OPC Package Release Version [e.g. 1.4.365.48]

**Additional context**
N/A

PaulHunkar, 2022-03-01:
The AliasNames that are described in part 17 - which is also where FindAlias method is defined, have nothing to with the Alias that is described in the nodeset.    The Alias in the nodeset are just used for the XML processing, i.e. an alternate name for something in the XML file.   They do not appear in the server as an AliasName.    The AliasNames that are described in part 17 allow an end user to generate an alternative name for something in the address space.  A typical use case might where a temperature has some name in the system like InletTemperatire, but it also has a name on the PID drawing of TI-101.  An AlliasName of TI-101 could be assigned to the InletTemperatire object.

andythinkiq, 2022-03-01:
Paul, thank you for the explanation. I was confused, but that totally makes sense now.

closed 2022-03-01 by andythinkiq