OPCFoundation/UA-.NETStandard #1732: Connect/listen via IP number

Light407, 2022-03-15:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [X] Help wanted

**Describe the Issue**

I am using the sample ReferenceServer as a base for my OPC-UA server and it works great. The problem I am facing now is that the endpoint the server registering is the host name of the computer, so if I connect to opc.tcp://<computername>:61234/ReferenceServer it works perfectly. However, in my setup usually the DNS/host lookup is not working and I have to be able to connect to the OPC-UA server directly with the IP number from the client, i.e opc.tcp://192.168.1.100:61234/ReferenceServer.

My problem is that this is not working, I am guessing it is because the client is rejecting the connection because the "hostname" is not matching. So my question is: how can I make sure the server is listening/accepting connections both via hostname and its IP number?

Thank you very much for the help

benjamin1988, 2022-03-16:
@Light407 

I'm not sure if i get it right, but i think you will have to add more endpoints at the server side. You first have to add the IP and second you have to add the endpoints via hostname. Then you should be able to connect via hostname and via IP address.

mregen, 2022-03-18:
Hi @Light407, 
we need to check if this may be caused by #1596, but I *think* it is ok if you specify the endpoint using 'localhost', it should bind to any ipaddress/hostname on the server.

you have multiple options to improve this situation:
i) if you have a static IP assigned or multiple hostnames to reach the server, adding these hostnames to the config adds them as valid hostnames to the server cert (if recreated):

```xml
    <AlternateBaseAddresses>
      <ua:String>http://AlternateHostName/Quickstarts/ReferenceServer</ua:String>
      <ua:String>http://10.10.103.150/Quickstarts/ReferenceServer</ua:String>
      <ua:String>http://[2a01::626d]/Quickstarts/ReferenceServer</ua:String>
    </AlternateBaseAddresses>
```

OR 

you allow to ignore the hostname on the client side. There is a 'check hostname' flag in `Session.Create` or a callback in the cert validator to allow to suppress the client hostname check error.

mregen, 2022-06-10:
please reopen if not answered

closed 2022-06-10 by mregen