OPCFoundation/UA-.NETStandard #1734: Version 1.4.368.33 session reconnection bug

xky0007, 2022-03-17:
**Type of Issue**
- [x] Bug 

**Describe the Issue**

SessionReconnectHandler won't reconnect session correctly and will never call Complete delegate in version 1.4.368.33.

**To Reproduce**
1. Update library from v1.4.367.100 to 1.4.368.33
2. Connect to a OPCUA server
3. Restart the server
4. Multiple sessions have been recreated but none can trigger session reconnect complete delegate.

**Expected behavior**
Only 1 session is re-created and Client_ReconnectComplete method can be triggered.

**Code snippets**
```
        private void Client_KeepAlive(Session sender, KeepAliveEventArgs e)
        {
            if (e.Status != null && ServiceResult.IsNotGood(e.Status))
            {
                if (reconnectHandler == null)
                {
                    if (m_ReconnectStarting != null)
                    {
                        m_ReconnectStarting(this, e);
                    }
                    Console.WriteLine($"{DateTime.UtcNow.ToString("yyyy-MM-ddThh:mm:ssZ")} {sender.Endpoint.EndpointUrl} reconnecting");
                    reconnectHandler = new SessionReconnectHandler();
                    reconnectHandler.BeginReconnect(sender, ReconnectPeriod * 1000, Client_ReconnectComplete);
                }
            }
        }
```

**Environment (please complete the following information):**
 - OS: [Windows 10]
 - Development environment: [e.g. Visual Studio 2022]
 - Runtime: [ .NET Core 3.1]
 - OPC Package Release Version [e.g. 1.4.368.33]

kimlukas, 2022-03-17:
This bug is also mentioned in https://github.com/OPCFoundation/UA-.NETStandard/issues/1734

xky0007, 2022-03-18:
> This bug is also mentioned in #1734

Hi, I think you linked wrong issue. #1734 is this post..

closed 2022-03-28 by mregen