OPCFoundation/UA-.NETStandard #1744: Opc.Ua.Core.dll file size

martinbu, 2022-03-21:
**Type of Issue**
- [x] Enhancement 
- [x] Question 

**Opc.Ua.Core.dll file size**

The OpcUa nuget was split up recently. But the `Opc.Ua.Core.dll` is still quite huge (> 8Mb). May it be possible to shrink this dll to some degree?

[image]

The `Opc.Ua.Core.dll` uses some resources. Are these resources really needed to be available in the core dll, or may it be possible to split them out or at least compress some of them, e.g. `NodeSet2.xml`?

[image]

mregen, 2022-03-26:
@martinbu the Nodeset2 xml had been requested by a user of the stack, I agree a zipped version would be nice. Just nobody had the time to investigate how to integrate zip in the build process. If you have some hints or time to provide a PR we are happy to take it.

martinbu, 2022-03-28:
@mregen I could have a look on how to integrate zipping into the build process.
But the other question is wether it is not easier and more lean to split the complete resources or at least the big once out of the core nuget, because I could not find a reference on NodeSet2.xml in the code base. But I may overlooked or missunderstood something.

Then the base users get a 50% smaller core nuget and the users needing the resources could add the dedicated nuget to get access to the NodeSet2.xml

PR #2232 (2023-07-27): Zip embedded Opc.Ua.NodeSet2.xml to reduce assembly size
## Proposed changes

- Add only the zipped NodeSet2.xml as embedded resource
- Reduces assembly size by at least 2MB
- Breaking Change: Embedded resource Opc.Ua.NodeSet2.xml is now Opc.Ua.NodeSet2.xml.zip

## Related Issues

- Fixes #1744 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [ ] Bugfix (non-breaking change which fixes an issue)
- [x] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement [... cut]
closed 2023-07-27 by mregen