OPCFoundation/UA-.NETStandard #1748: ConsoleReferenceServer exception

sergear2ip, 2022-03-25:
**Type of Issue**
- [ ] Bug 
- [ ] Help wanted

consolereferenceserver .net4.6.2 binary causes Microsoft.Extensions.Primitives.InplaceStringBuilder exception at runtime 

 - OS: Windows 10
 - Development environment: Visual Studio 2019
 - Runtime: NET 4.6.2

mregen, 2022-03-26:
@sergear2ip could you provide more info, e.g. callstack? Just a guess, may have to do with logging because its in Microsoft.Extensions...

sergear2ip, 2022-03-26:
I added console.readline() at the end of main of the ConsoleRerenceServer native application main to visualize the stacktrace below 

.NET Core OPC UA Reference Server
Loading configuration from Quickstarts.ReferenceServer.
Check the certificate.
Start the server.
Application startup exception: System.TypeLoadException: Impossible de charger le type 'Microsoft.Extensions.Primitives.InplaceStringBuilder' à partir de l'assembly 'Microsoft.Extensions.Primitives, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60'.
   à Microsoft.Net.Http.Headers.DateTimeFormatter.ToRfc1123String(DateTimeOffset dateTime, Boolean quoted)
   à Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.DateHeaderValueManager.SetDateValues(DateTimeOffset value)
   à Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServer.CreateServiceContext(IOptions`1 options, ILoggerFactory loggerFactory)
   à Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServer..ctor(IOptions`1 options, ITransportFactory transportFactory, ILoggerFactory loggerFactory)
--- Fin de la trace de la pile à partir de l'emplacement précédent au niveau duquel l'exception a été levée ---
   à System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   à Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.VisitConstructor(ConstructorCallSite constructorCallSite, RuntimeResolverContext context)
   à Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteVisitor`2.VisitCallSiteMain(ServiceCallSite callSite, TArgument argument)
   à Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.VisitRootCache(ServiceCallSite callSite, RuntimeResolverContext context)
   à Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteVisitor`2.VisitCallSite(ServiceCallSite callSite, TArgument argument)
   à Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.Resolve(ServiceCallSite callSite, ServiceProviderEngineScope scope)
   à Microsoft.Extensions.DependencyInjection.ServiceProvider.CreateServiceAccessor(Type serviceType)
   à System.Collections.Concurrent.ConcurrentDictionary`2.GetOrAdd(TKey key, Func`2 valueFactory)
   à Microsoft.Extensions.DependencyInjection.ServiceProvider.GetService(Type serviceType, ServiceProviderEngineScope serviceProviderEngineScope)
   à Microsoft.Extensions.DependencyInjection.ServiceProviderServiceExtensions.GetRequiredService(IServiceProvider provider, Type serviceType)
   à Microsoft.Extensions.DependencyInjection.ServiceProviderServiceExtensions.GetRequiredService[T](IServiceProvider provider)
   à Microsoft.AspNetCore.Hosting.Internal.WebHost.EnsureServer()
   à Microsoft.AspNetCore.Hosting.Internal.WebHost.BuildApplication()
The application exits with error: Impossible de charger le type 'Microsoft.Extensions.Primitives.InplaceStringBuilder' à partir de l'assembly 'Microsoft.Extensions.Primitives, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60'.

mregen, 2022-03-28:
Hi @sergear2ip, thanks for bringing this up. There are some incompatibilites between some Microsoft.Extensions nuget packages and the ASP.NET 2.1 kestrel server used in .NET4.6.2 references an older version that is incompatible with the latest.
One way to fix this is by manually referencing Extensions.Primitives and logging in the app, but I think the better solution is to reference the older versions in core.

sergear2ip, 2022-03-28:
Hi @mregen,
I looked this weekend, indeed the "InplaceStringBuider" function is announced as obsolete by Microsoft.
Thanks for your help

closed 2022-03-28 by sergear2ip