OPCFoundation/UA-.NETStandard #1750: JWT Tokens throws NotSupportedException

Zigby, 2022-03-28:
**Type of Issue**
 Bug 

**Describe the Issue**
UserIdentity.cs (server side) throws an NotSupportedException when a client passes a JWT token as IssuedIdentityToken.

``` c#
new NotSupportedException("Only JWT Issued Tokens are supported!");
``` 

The problem is that the IssuedTokenType property is never set, this property is not serialized over the wire and needs to be set by the server SDK, but there is no such assignment.
On the wire the user token policy is sent and it is probably the user token that needs to be checked (the IssuedTokenType property could be set based on the value of the policy.

**To Reproduce**
Steps to reproduce the behavior:
*Connect to a server built with this SDK, with support for the JWT tokens in the config xml file.
* Pass a JWT token as an IssuedIdentityToken from a client (should not matter what client, I used one built using this SDK).
=> Exception is thrown in UserIdenty.cs on the server.

**Expected behavior**
A UserIdentity object is sucessfully created with the JWT token..

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log output**
If applicable, add log output to help explain your problem.

**Code snippets**
See UserIdentity.cs:

``` C#
IssuedIdentityToken issuedToken = token as IssuedIdentityToken;
if (issuedToken != null)
{
    if (issuedToken.IssuedTokenType == Ua.IssuedTokenType.JWT) // this will always fail as the IssedTokenType is never assigned
    {
        if (issuedToken.DecryptedTokenData == null || issuedToken.DecryptedTokenData.Length == 0)
        {
            throw new ArgumentException("JSON Web Token has no data associated with it.", nameof(token));
        }

        m_tokenType = UserTokenType.IssuedToken;
        m_issuedTokenType = new XmlQualifiedName("", Opc.Ua.Profiles.JwtUserToken);
        m_displayName = "JWT";
        return;
    }
    else
    {
        throw new NotSupportedException("Only JWT Issued Tokens are supported!");
    }
```

**Environment (please complete the following information):**
 - OS: Windows 11
 - Development environment: Visual Studio 2022 17.1.1]
 - Runtime: .NET 6.0.201
 - OPC Package Release Version: 1.4.368.33
 - Component:  Opc.Ua.Core\Stack\Client
 - Server: e.g. Reference Server]
 - Client: Client with support for JWT

**Additional context**
Automatic tests for JWT seem to be missing, adding a new test case could be a good way to reproduce the problem.

Zigby, 2022-04-04:
Hi @mregen, 
the PolicyId is transferred over the wire. On the receiver side the token policy is looked up by PolicyId (index), and as far as I can tell the IssuedTokenType needs to be set on the receiver side based on the looked-up policy. 
Like this:

```C#
Opc.Ua.Server/Session/Session.ValidateUserIdentityToken(..)
{ 
... 
  if (token is IssuedIdentityToken issuedToken)
  {
     if (policy.IssuedTokenType == Profiles.JwtUserToken)
     {
         issuedToken.IssuedTokenType = IssuedTokenType.JWT; 
     }
 }
...
```

Perhaps there is a better way to do it but the PolicyId is the only field transferred over the wire that "contains" the information.

mregen, 2022-04-04:
Hi @Zigby, now it makes sense, the server must define the policy for the JWT token, only then the client can send the JWT token and 'pass' the check.

Zigby, 2022-04-27:
Hi @mregen, 
a change in the server side code, something like I suggested, is still required in order to get the IssuedTokenType as JWT instead of as IssuedTokenType.GenericWSS (default enum value). 

Workaround currently is to ignore the token type in the server and just assume it is a JWT anyways. Setting the token type in the ValidateUserIdentityToken method as suggested would be preferred, or maybe there is some other better solution. Currently I get the wrong type in my server, and I cannot find anywhere in the source where the IssuedTokenType is being assigned, hence leaving the IssuedTokenType with the default value GenericWSS. 

Should I make a pull request with the proposed change, so you can review? 
It is the following lines of code that I have added to the Session.ValidateUserIdentityToken method:

```C#
Opc.Ua.Server/Session/Session.ValidateUserIdentityToken(..)
{ 
... 
  if (token is IssuedIdentityToken issuedToken)
  {
     if (policy.IssuedTokenType == Profiles.JwtUserToken)
     {
         issuedToken.IssuedTokenType = IssuedTokenType.JWT; 
     }
  }
...
```

mregen, 2022-05-08:
> Should I make a pull request with the proposed change, so you can review?

yes please!

mregen, 2022-06-29:
@Zigby, I wonder if here the XmlQualified name is not correctly initialized for the encoding, the JWT user token should be first not the namespace?

m_issuedTokenType = new XmlQualifiedName("", Opc.Ua.Profiles.JwtUserToken);

closed 2022-06-30 by mregen