OPCFoundation/UA-.NETStandard #1766: <title>

utthungagokul, 2022-03-31:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

System.Formats.Asn1.AsnContentException
  HResult=0x80131500
  Message=The provided data is tagged with 'Universal' class value '13', but it should have been 'Universal' class value '16'.
  Source=System.Formats.Asn1
  StackTrace:
   at System.Formats.Asn1.AsnDecoder.CheckExpectedTag(Asn1Tag tag, Asn1Tag expectedTag, UniversalTagNumber tagNumber)
   at System.Formats.Asn1.AsnDecoder.ReadSequence(ReadOnlySpan`1 source, AsnEncodingRules ruleSet, Int32& contentOffset, Int32& contentLength, Int32& bytesConsumed, Nullable`1 expectedTag)
   at System.Formats.Asn1.AsnReader.ReadSequence(Nullable`1 expectedTag)
   at Opc.Ua.Security.Certificates.X509Signature.Decode(Byte[] crl) in \Libraries\Opc.Ua.Security.Certificates\X509Crl\X509Signature.cs:line 132

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2019
- Runtime: .NET Core 3.1
- Nuget Version:
- Component: Opc.Ua.Core
- Server: Reference Server
- Client: Reference Client
```

### Anything else?

I'm trying to run the Reference Server with CA signed certificate with Store Type of Directory 
Following are the structure of the certificates 
%CommonApplicationData%\{ApplicationName}\pki\own\certs\certificate.der --> public key
%CommonApplicationData%\{ApplicationName}\pki\own\private\certificate.pfx --> private key

%CommonApplicationData%\{ApplicationName}\pki\trusted\certs\intermediateRootCa.der --public key
%CommonApplicationData%\{ApplicationName}\pki\trusted\certs\rootCa.der --public key

%CommonApplicationData%\{ApplicationName}\pki\trusted\crl\rootca.crl --> Certificate Revocation List

when add the CRL file always throw's the message's The provided data is tagged with 'Universal' class value '13', but it should have been 'Universal' class value '16'.
 
Need the solution for this case.

mregen, 2022-04-01:
@utthungagokul please provide the certificates you can share, specifically the crl. Looks like the ASN.1 encoding of the certificate is not accepted because it doesn't start with a sequence.

utthungagokul, 2022-04-01:
@mregen Thanks for the replay,

Below is the CRL file 

X509 Certificate Revocation List:
Version: 2
Signature Algorithm:
    Algorithm ObjectId: 1.2.840.113549.1.1.11 sha256RSA
    Algorithm Parameters:
    05 00
Issuer:
    CN=..
    OU=..
    O=..
    L=..
    S=..
    C=..
  Name Hash(sha1): 16f5c86fa388d8d0767124d8e52f0be3d674aecf
  Name Hash(md5): 4f2bc92d80845589c22a5a302211a723

 ThisUpdate: 2022-04-01 04:26 PM
 NextUpdate: 2022-05-01 04:26 PM
CRL Entries: 1
  Serial Number: 3bbe82ce2e28c9e57f663066063df959e4515b6e
   Revocation Date: 2022-03-31 10:11 PM

CRL Extensions: 2
    2.5.29.35: Flags = 0, Length = b5
    Authority Key Identifier
        KeyID=14badc0a23e3fbf222789a90c0242ec875d674ca
        Certificate Issuer:
             Directory Address:
                  CN= ..
                  OU=..
                  O=..
                  L=..
                  S=..
                  C=..
        Certificate SerialNumber=3bbe82ce2e28c9e57f663066063df959e4515b6e

    2.5.29.20: Flags = 0, Length = 4
    CRL Number
        CRL Number=1004

Signature Algorithm:
    Algorithm ObjectId: 1.2.840.113549.1.1.11 sha256RSA
    Algorithm Parameters:
    05 00
Signature: UnusedBits=0
    0000  9b 18 7c 65 e4 5f 15 54  29 fe cd 70 04 13 1c 65
    0010  c1 ee 55 85 87 83 4a f5  cf 18 6f ba fb 46 f0 6b
    0020  f9 a2 2d c6 34 2d 47 65  8d ce 3c 7b 09 1d c6 81
    0030  5e e6 22 5d 7c 15 3a bb  70 1f 3b 8a 63 f5 87 ea
    0040  37 10 00 ba b1 51 6e af  7b db 65 aa 17 75 a7 65
    0050  c7 46 8b fc 8b 7b 82 a0  1c e2 5d 9b 74 9f c5 b1
    0060  3e b3 39 89 fa f6 d5 18  e1 74 27 49 80 09 45 a2
    0070  19 cf b2 a3 a2 a3 91 f3  cd db e0 b1 0f a8 51 d2
    0080  76 fe ce f0 75 a0 06 5f  36 1a e4 6b 23 3a d8 03
    0090  69 25 1c 35 dc 8d 05 3e  e0 39 64 a1 52 52 51 c4
    00a0  7f 0e e2 70 ce 2a 9e 64  48 b5 77 a7 3f 93 14 b7
    00b0  ae 7a 8f 1f de 9e a0 bb  57 20 b3 6f 7e 2d 4a fa
    00c0  21 86 2a 6a b6 3e 14 9b  9b 04 d4 84 a6 99 d8 33
    00d0  fd 20 20 ae fe c3 7c 80  f5 2e ec 20 f7 de 95 12
    00e0  87 9c 22 4b 1f 9d ea 4a  52 e5 1d 38 bc 99 18 e6
    00f0  dd 3d 9a aa 7a b6 1c f3  73 1d 76 7b dc a7 d7 91
    0100  42 aa 07 b2 96 49 88 23  40 c0 08 ca e5 e9 50 1f
    0110  18 b8 32 22 5c 61 5e 16  99 8f 8c 90 7f de 61 ca
    0120  b8 c7 48 6f fb 82 23 f0  44 1e 6c b1 6b c2 f0 f6
    0130  f7 10 25 54 48 bd 84 f6  a1 8c 58 92 81 ce 92 71
    0140  a0 95 e0 79 70 52 b5 12  37 a1 cf fa a4 4e 9e fb
    0150  d3 f9 b4 fd f9 dd 21 51  16 4e 9f a4 b7 f1 35 8d
    0160  20 52 c6 f4 44 e8 2e 50  e0 b4 32 bf 3e dd c8 93
    0170  4c ff c1 02 39 34 ab 5f  ed 0e 2f 25 67 a7 48 5f
    0180  76 7a fa 5c 77 6e df 33  11 ca 37 68 8e 9e 95 8d
    0190  2c 47 43 80 33 b8 1e df  52 d1 13 ae f2 be ce 8e
    01a0  4b e5 27 f5 a2 dd 2a 44  b6 65 56 9f bd d7 6a 80
    01b0  7a 8d 02 f0 5e cf 45 54  89 51 f2 b7 f7 50 19 36
    01c0  bd e7 d0 bd fa af 22 71  3d 61 e8 b3 51 17 78 61
    01d0  37 13 71 78 42 5a b3 c6  2d d1 28 73 30 56 fb ef
    01e0  51 64 4b 8b c9 a9 bc 31  cf 4e 84 da a6 f3 25 71
    01f0  f4 3b 24 59 fb ad ff ce  d4 92 27 53 dd c8 84 e8
CRL Hash(md5): ba7d7e28a227ba5ccf401a15cbb7adc8
CRL Hash(sha1): 66e79c6214b85c2c17d3af0da23c8b095dcf9037
CRL Hash(sha256): 765006d596752dd6bd32861c6387f81d264db95f1d16082c6ed864f9629cc633
Signature Hash: 27c6d8a1eae001b5e2619423157e6bc85cdfbe999dc51a5e8da920a3210a9f9c
CertUtil: -dump command completed successfully.

mregen, 2022-04-01:
Hi @utthungagokul, I would prefer the binary blob of the CRL. I guess you can attach a zip here.

utthungagokul, 2022-04-04:
Hi @mregen ,

I have attached the file for the support, Kindly have a look on that
[pki.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/8407516/pki.zip).

mregen, 2022-04-06:
Hi @utthungagokul , the CRL is in the unexpected PEM format, but the cert store expects a CRL in ASN.1 encoding.  -->
```
-----BEGIN X509 CRL-----
MIIDxDCCAawCAQEwDQYJKoZIhvcNAQELBQAwfjELMAkGA1UEBhMCSU4xCzAJBgNV
...
1sdrBMRCDVqLnS9dLxuBaqyrJ/bG7mzMNrF/CKH9deeAZ/B7UjE06g5UBXxUci/C
lbx/F6nK3J0=
-----END X509 CRL-----
```

Now the question is how common is your use case and should it be 'auto detected' by the CRL decoder?

utthungagokul, 2022-04-07:
Hi @mregen 

My use case is to use the CA signed certificate instead of the auto generated  Self-Signed certificate.
To achieve I have create RootCA, Intermediate CA , and Application Server signed by intermediate and placed the certificate into the Local Certificate Store in the Directory as I mentioned in the Question. Later come to know the CRL file is expected in the trusted stored. So that I have created the CRL file using the OpenSSL tool and placed the CRL files into trusted stored.
After the the all these steps got into the CRL encoding issue. 
I not sure that CRL should auto detected or not. Here I need a further help.

Would be great helpful if suggested the offline tools to Manage the Certificate Authority's.

mregen, 2022-04-08:
> Would be great helpful if suggested the offline tools to Manage the Certificate Authority's.

The Opc.Ua.NETStandard.Security.Certificates Nuget library contains all necessary C# code to manage certificates and CRLs. 
For OPC UA applications a GDS can be used. A sample application is in the samples repo [here](https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/Samples/GDS).

utthungagokul, 2022-04-11:
Thanks you @mregen , I will look into the the GDS application.

mregen, 2022-04-19:
At this point I believe the issue can be closed, please feel free to reopen.

mregen, 2022-09-26:
consider to support PEM encoded CRL

mregen, 2022-10-24:
one more... https://github.com/Azure/Industrial-IoT/issues/1886

closed 2023-01-30 by utthungagokul