OPCFoundation/UA-.NETStandard #1772: How to write an ExtensionObject?

lmioco, 2022-04-06:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

Hello everyone, 
I can't figure out how to write an extension object, I always receive `BadTypeMismatch` status code as result.  
Using the `BinaryEncoder` I create a `byte[]` that contains my object data.  
I tried several different things to valorize Value for example:

```
Value = new DataValue(new ExtensionObject(binaryData))
or
Value = new DataValue(new Variant(new ExtensionObject(binaryData)))
```

Below there is a portion of the code used to write
``` csharp
var valuesToWrite = new WriteValueCollection()
    {
         new WriteValue()
         {
             AttributeId = Attributes.Value,
             NodeId = new NodeId(ctx.WriteDataNode),
             Value = // What goes here?
         }
    };

opcua.Session.Write(
    null,
    valuesToWrite,
    out results,
    out _);
```

### Environment

```markdown
- OS: Windows 10
- Environment: Visual studio 2022
- Runtime: net6.0
- Nuget Version: Latest
- Component: Opc.Ua.Core
- Server: PLC SIEMENS
```

### Anything else?

This is the node I want to write
[image]

And this is how I encode it:
``` csharp
var encoder = new BinaryEncoder(ServiceMessageContext.GlobalContext);
encoder.WriteBoolean("", PcData.PcOK);
encoder.WriteBoolean("", PcData.EndOfCheckOK);
encoder.WriteBoolean("", PcData.EndOfCheckKO);
encoder.WriteBoolean("", PcData.EnableTransit);
encoder.WriteBoolean("", PcData.EnableRecive);
encoder.WriteBoolean("", PcData.CumulativeError);
encoder.WriteByte("", PcData.MicrocycleRecipe);
encoder.WriteByte("", PcData.ErrorCode);

return encoder.CloseAndReturnBuffer();
```

lmioco, 2022-04-06:
I now tryed to write the DataValue readed from the same node, and the result was `BadWriteNotSupported`.  
Even if the access level on the node is CurrentWrite
[image]

If you don't have any advice I will close the issue at this point.

mrsuciu, 2022-04-08:
@Imioco 
I think that what you are trying to do fails because you are trying to write a new value to a complex data type simply by wrapping the binary encoding into an ExtensionObject. By doing this the type Info is skipped and BadTypeMismatch  is returned.
Probably an example of how to encode a complex data type will help (ex EncodeDecodeComplexType under Opc.Ua.ClientComplexTypes.Tests.Types.ComplexTypesCommon)

closed 2023-07-07 by EthanChangAED