OPCFoundation/UA-.NETStandard #1773: OPCUA client refuses connection with OPCUA server in none-secured mode if the server send wrongly its certificate.

GregoireG-C, 2022-04-07:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

As the title indicates. This happens because the OPCUA client checks the Server certificate and can refuses it (if the certificate is untrusted or for other reasons (key size, date validity)).

You will notice that this happen with a non-compliant server.

Extracted from OPCUA standard Part4 §5.4.3.1 : _"If the securityPolicyUri is NONE and none of the UserTokenPolicies requires encryption, the Server shall not send an ApplicationInstanceCertificate and the Client shall ignore the ApplicationInstanceCertificate."_

This means that the standard impose OPCUA client to accept connexion with none-compliant server (probalby for upward compatibility).
 

### Expected Behavior

The client should accept connection with servers in None-secured mode if the server send a bad certificate which is not compliant with OPCUA standard. 

### Steps To Reproduce

1. Modify a compliant server to make it send its certificate in none-secured connexion.
2. Connect OPCUA client to it in none secured mode. => The client should refuse the connection.

### Environment

```markdown
- OS:Windows 10
- Environment:Visual Studio 2019 16.11.3
- Runtime: .Net 4.8
```

GregoireG-C, 2022-04-07:
The code impacted is located in Libraries.Opc.Ua.Client/Session : public void Open(
            string sessionName,
            uint sessionTimeout,
            IUserIdentity identity,
            IList<string> preferredLocales,
            bool checkDomain)

romanett, 2024-05-06:
Can you check again with the current library version?
If the issue is really in the Open call to me it seems it is resolved, as we only check the server certificate if the security policy is not "none":

https://github.com/OPCFoundation/UA-.NETStandard/blob/663ed10285153955bc009e464fb35f162512aaa7/Libraries/Opc.Ua.Client/Session.cs#L2320

closed 2024-07-08 by romanett