OPCFoundation/UA-.NETStandard #1776: Hi @MD-V,

crocodile20, 2022-04-13:
Hi @MD-V, 
on the machine I'm using you have to create what is called "protocol" from the panel of the machine, then within this "protocol" I'm going to insert the nodes that interest me. I do not know if I can share the documentation because I can not find it online. 

However to give an idea this is a screenshot of ua expert with the simulator server, it is not the same but the object highlighted is similar to the real situation (now I can not take a screenshot of that because I do not have access but as soon as I can I'll add that). 

[image]

In the example screenshot inside "TransitionEventType" there are Transition, FromState and ToState, while in the real situation I see all the nodes I need to read, i.e. those inserted in the "protocol"

_Originally posted by @matteomessmer in https://github.com/OPCFoundation/UA-.NETStandard/issues/1429#issuecomment-892745264_

matteomessmer, 2022-04-14:
Hi @crocodile20, I found this online https://www.euromap.org/euromap77
The documentation I had was the one the manufacturer gave me, but it wasn't very helpful for developing the client.
What I did was exactly as benjamin wrote here https://github.com/OPCFoundation/UA-.NETStandard/issues/1429#issuecomment-900333321
I browse the nodes starting from a known node ID, searching the nodes that I want to read (you have to create a list of those nodes names, for doing this you can use UAExpert). When you have found all of them you have all the IDs and you can read all the nodes values. 
You can use the free OPC UA server simulator https://prosysopc.com/products/opc-ua-simulation-server/ to test your client

benjamin1988, 2022-04-14:
Hi @crocodile20,

alarms are just events of a specific type. So you have to create a event subscription to receive the information. https://github.com/OPCFoundation/UA-.NETStandard/issues/1429#issuecomment-903830900 contains the information you need. You only have to know that the EM77 alarms are just AlarmConditionType Events and not the standard UA Alarms.
If you need further information feel free to ask. 

With kind Regards
Benjamin

crocodile20, 2022-04-14:
Thank you 🙏 I will try this method and come back to you

crocodile20, 2022-05-09:
> Hi @benjamin1988 
I have a question about UA expert 
I have a function in Ua expert to do a call for alarms 
[image]

[image]

[image]

how to get the right node id to visualize the alarms

benjamin1988, 2022-05-10:
Hello @crocodile20,

I'm not sure what you are searching for? You can find the detailed node infromation in the attribute view of UA Expert:
[image]
In our application we first browse the whole address space, afterwards we are working with the Browse Names by translating them from Browse Name into Node id. Therefore we save the browsed information into a file or dictionary.

With kind Regards,
Benjamin

EthanChangAED, 2023-06-21:
https://github.com/OPCFoundation/UA-.NETStandard/issues/1429

closed 2023-06-21 by EthanChangAED