OPCFoundation/UA-.NETStandard #1779: Application Certificate and Trusted Peer Certificates

DmitryFarberov, 2022-04-21:
How come an Application Instance certificate is required to be present in the "Trusted Peer Certifcates" store (see callstack)? I would've thought that just being in the "Application Certificate" store should be sufficient?

Thanks.

Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.GetTrustedCertificate(System.Security.Cryptography.X509Certificates.X509Certificate2 certificate) Line 547
	at D:\Sourcetree\uap_complete\OPC UA\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs(547)
Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.InternalValidate(System.Security.Cryptography.X509Certificates.X509Certificate2Collection certificates, Opc.Ua.ConfiguredEndpoint endpoint) Line 877
	at D:\Sourcetree\uap_complete\OPC UA\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs(877)
Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.Validate(System.Security.Cryptography.X509Certificates.X509Certificate2Collection chain, Opc.Ua.ConfiguredEndpoint endpoint) Line 366
	at D:\Sourcetree\uap_complete\OPC UA\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs(366)
Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.Validate(System.Security.Cryptography.X509Certificates.X509Certificate2Collection chain) Line 351
	at D:\Sourcetree\uap_complete\OPC UA\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs(351)
Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.Validate(System.Security.Cryptography.X509Certificates.X509Certificate2 certificate) Line 327
	at D:\Sourcetree\uap_complete\OPC UA\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs(327)
Opc.Ua.Configuration.dll!Opc.Ua.Configuration.ApplicationInstance.CheckApplicationInstanceCertificate(Opc.Ua.ApplicationConfiguration configuration, System.Security.Cryptography.X509Certificates.X509Certificate2 certificate, bool silent, ushort minimumKeySize) Line 604
	at D:\Sourcetree\uap_complete\OPC UA\UA-.NETStandard\Libraries\Opc.Ua.Configuration\ApplicationInstance.cs(604)
Opc.Ua.Configuration.dll!Opc.Ua.Configuration.ApplicationInstance.CheckApplicationInstanceCertificate(bool silent, ushort minimumKeySize, ushort lifeTimeInMonths) Line 452
	at D:\Sourcetree\uap_complete\OPC UA\UA-.NETStandard\Libraries\Opc.Ua.Configuration\ApplicationInstance.cs(452)
Opc.Ua.Configuration.dll!Opc.Ua.Configuration.ApplicationInstance.CheckApplicationInstanceCertificate(bool silent, ushort minimumKeySize) Line 400
	at D:\Sourcetree\uap_complete\OPC UA\UA-.NETStandard\Libraries\Opc.Ua.Configuration\ApplicationInstance.cs(400)

mregen, 2022-05-06:
Hi @DmitryFarberov, this requirement was addressed in the 367 release timeframe by ignoring suppressible errors on startup. see #1794

mregen, 2022-06-10:
please reopen if not answered

closed 2022-06-10 by mregen