OPCFoundation/UA-.NETStandard #1781: What does KeepAlive handle?

gurc4n, 2022-04-23:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I have an MES application, developed internally.
Using Kepware as the OPC server.

I have disconnection issues occasionally. I have tried the KeepAlive and Client Reconnect logic in the sample application. But e.Status is null at the end, and it won't reconnect.

I have implemented my own logic as follows;
1) query server time every 500 milliseconds.
2) if the gap between the current time and the server time exceeds a certain limit, I simply dispose everything and start from scratch.

As you can imagine this is not a preferred way, and it takes some time to reconnect. But the production machines require real-time data transfer. So every reconnection means downtime for production.

My question is:
What am I doing wrong with KeepAlive? Is there a special trick to make it work?
1) Will it reconnect if there is a network issue? (such as someone taking the network cable out of the server)?
2) Will it reconnect after a kepware reinitialization?
3) Will it reconnect if I get an error message as "Server closed connection"?

In other words, Does handling KeepAlive and ReconnectHandler solve all reconnection-related issues? Or Should I do something else?

### Environment

```markdown
- OS: Windows Server 2016
- Environment: Visual Studio 2022
- Runtime: .NET 4.8
- Nuget Version: 1.04.368
```

mrsuciu, 2022-04-26:
@gurc4n

The KeepAlive-Reconnect logic should detect the communication errors produced by the steps you've mentioned.

 As I see, the only places that `KeepAliveEventArgs.Status` could be set to `null ` is either if KeepAliveEventHandler@ `Opc.Ua.Client.Session.OnKeepAlive(ServerState currentState, DateTime currentTime) Line 4170` is called through `Opc.Ua.Client.Session.ProcessPublishResponse(...) Line 4580` or 
through `Opc.Ua.Client.Session.OnKeepAliveComplete(System.IAsyncResult result) Line 4109`.
In both cases the `Opc.Ua.Client.Session.OnKeepAlive(ServerState currentState, DateTime currentTime)` is called when the server returns a keep alive response, therefore the connection with the server should be active at that moment, therefore reconnecting should not be needed in that case.

Could it be that you have a "buggy" setup for the Opc.Ua.Client.Session.KeepAlive event handling mechanism?
Did you try to follow the way `Opc.Ua.Client.Controls.ConnectServerCtrl.Session_KeepAlive(Session session, KeepAliveEventArgs e)` is set up as event handler for the Opc.Ua.Client.Session.KeepAlive event (implemented under the UA Client Controls project) ?

benjamin1988, 2022-04-26:
Hello,

i have a similair problems with a siemens opc ua server. First I have implemented a reconnection on my own by using the Reconnect() function of the stack in combination with a timer.
Now I have changed the reconnection to the SessionReconnectionHandler and the KeepAlive Status, until know without any problems. But after a reconnection we have problems with the subscriptions.

1. We first have tried to add the subscriptions again after reconnection, without any success. 
2. Now we try to create new subscriptions with the same parameters but also with no success.

Questions:
- Do subscriptions have to be renewed after the reconnectoin via Session Reconnection Handler?
- How can i check if the server supports durable subscriptions (as I could see the support of this would solve this)?

Is it possible that the SessionReconnectionHandler could only solve some connection problems as @gurc4n already said? And that their is a point in the program, where only the creation of a new session solves the problem? 

With kind Regards
Benjamin

gurc4n, 2022-04-26:
Hi @mrsuciu ,

I have followed and implemented precisely the same as in the sample applications.

_I am unsure about the best practices but let me summarize;_
There are 10 production lines.
I create a session for each production line.
There are 20 stations per production line, I create one subscription per line, and there are 2 tags that I need to read and respond to on each station. So I added those 2 tags as monitored items. There are 40 monitored items per production line.
There are some exceptional cases like, I have 1000 tags to monitor, I create another subscription for them. And it doesn't hurt in a reconnect sequence.

I started those sessions in a loop and was trying to subscribe to the Opc.Ua.Client.Session.KeepAlive, just after creating the session object. I see that during the other sessions were trying to connect there is a strange thing happening with KeepAlive. I was not able to connect, I had so many errors such as SubscriptionIdInvalid etc.

I have changed the event handler registration time to "after **all sessions** are created," and it is good now.
But again; I have tried to unplug the network cable and replug... and the test failed.

**What should be the best practice here?**
a) Current: 1 session = 1 production line, 1 session = 1 subscription
b) 1 session = all production lines, 1 subscription for all monitored items.
c) 1 session = all production lines, multiple subscriptions (1 for each production line)

Is there a maximum number of monitored items that I need to worry about?

benjamin1988, 2022-04-26:
Hi @gurc4n ,

the issue with the SubscriptionIdInvalid I had too. You could solve it by changeing the handler registration aft all sessions are create, how did you mean that?
Are the 10 production lines at the same OPC UA Server?

May I can help you a little bit with your questions:
If I'm right the maximum number of monitored items are described by the server. At the Server Object there is a node ServerCapabilities and there you can find the folder Operations Limits with the MaxMonitoredItemPerCall. [https://reference.opcfoundation.org/v104/Core/docs/Part5/6.3.11/](url)
[image]

The best practice should be simple, as the answer c already says that all the production lines are at the same opc ua server target, i think this should be the best solution. The only bottleneck den is the application itself. 
Otherwise you may also have problems with the maximum numbers of current sessions. 

Kind regards
Benjamin

gurc4n, 2022-04-27:
Hi @benjamin1988 

I will try to explain my logic:
1) Service starts
2) Reads the production line list.
3) Starts the production lines in a for each loop.
4) Every production line has its own wrapper for OPC UA connection. Therefore each of them creates another session and subscription etc.

Previous state:
Immediately after creating the session object, I was registering the KeepAlive handler.
Current state:
I have waited for all sessions to be created and registered the KeepAlive handler for each session in another loop.

Yes, I have only one UA server (Kepware server).

And sadly, Kepware doesn't have that OperationLimits folder in ServerCapabilities.

In my case, KeepAlive can re-establish the connection for example if I put a debug point and hang the application intendedly. 
It will reconnect immediately after I remove the debug point.

But It won't handle if I take the network cable out for some time and plug it in.

Regards,
Gurcan

benjamin1988, 2022-05-02:
Hi @gurc4n,

thanks for you explanations. 
I have found something with descripes what to do step by step to reconnect and reestablish subscriptions. May this helps to find a solution for your issue. [https://reference.opcfoundation.org/v104/Core/docs/Part4/6.7/](url) May you can take a look there. I will try to rebuild this sequence in my code and tell if it's working. 
It specially shows what to do if the Error Bad_SubscriptionIdInvalid is thrown.

With kind Regards
Benjamin

clane2812, 2022-05-05:
Hello - i can confirm that the client-reconnect-handling broke down between the version 1.4.367.100 and the version 1.4.368.33:

behaviour with 1.4.367.100:

- session with subscription established
- disconnect/connect VPN (think pulling the cable is the same)
- keep-alive-handler is called fastly (with the given interval) and status 'BadNoCommunication/Server not responding to keep alive requests.'
- within the keep-alive-handler 'BeginReconnect' is called on the reconnectHandler of the session
- the reconnect-handler-callback is called after some seconds
--> everything fine, the new subscriptions are automatically created and data flows in again!

behaviour with 1.4.368.33 (and greater):

- session with subscription established
- disconnect/connect VPN (think pulling the cable is the same)
- keep-alive-handler seems to hang and is called after exactly 120sec (no matter which keep-alive-timeout is set!). Status is 'BadNoCommunication/Server not responding to keep alive requests.' again.
- after calling the 'BeginReconnect' the reconnectHandler is NEVER called!
- instead of this every ~10-12sec the keep-alive-handler is called again, passing in a new session-object in every-call. The status of the session is 'Running', no subscription is attached to the new session
--> no more data flow

so - i fall back to the 1.4.367.100 package, till it´s confirmed or solved if this is a bug (i guess) or intended behaviour

mregen, 2022-05-06:
Hi @clane2812, first issue in 368 is #1783.
Second I'm investigating there is a regression in Session reconnect. Investigating.

mregen, 2022-05-08:
see #1802 for sample

benjamin1988, 2022-05-10:
Hi @gurc4n,

I'm just back after almost a week of tests of reconnection with a Siemens OPC UA Server. First of all the reconnection with the reconnection handler seems to work very well. We just had some issues with the subscriptions. Therefore we had to adapt our client dll to delete all subscriptions at the OnSessionClosing event. Delete the subscriptions means, that we first remove all monitored items then we remove the subscription and after removing all monitored Items and subscriptions we dispose the objects.
After sucessfully reconnect we get a new session which again all subscriptions. As we handle that like a new connection we first again have to remove all monitored items and subscription to be sure that we start from scratch. As the specific Siemens OPC has the limitation of max 1000 monitored items at all, we have to be sure that it is working, otherwise we aren't able to renew our subscriptions.

May i could help you a little bit.

With Kind regards
Benjamin

gurc4n, 2022-05-11:
Hi @benjamin1988 ,

That's exactly what I have tried in a previous version of my application.
The logic was similar to the below:
1) Create connections and subscriptions for each production line.
2) Monitor OPC/UA server time with a loop if the time gap is bigger than 5 seconds. Dispose everything and start from scratch. 

But that ends up with a production downtime in my case as it takes like 15-20 seconds.

That's why I wanted to try the library's reconnect handler, but couldn't achieve the expected result.

I have tried the same as you have mentioned in your post one more time but the **SessionClosing event doesn't fire on reconnect stage**.

So I changed the DeleteSubscriptionsOnClose event to true and fire an event to renew subscriptions after reconnect completed.
on that event, I have tried to delete the current subscription one more time, and created the subscriptions again from scratch.

I hope the new version that @mregen is working on will fix the issues.

mregen, 2022-05-14:
Hi @gurc4n and @benjamin1988 , please test the fixes to reconnect.
See #1802 for issues found and fixed.
There is also *best practice* to implement session reconnect here: https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/ConsoleReferenceClient/UAClient.cs
The subscription transfer on reconnect is now opt-in. Apparently the session and subscription lifetime need to be setup accordingly for transfer to work.

clane2812, 2022-05-16:
@mregen - Hi, i tested the 1.4.368.58 package and the reconnect-handling works again. Thanks for your work!

gurc4n, 2022-05-16:
I am running the test version on the test server. I will update you in a couple of days.

gurc4n, 2022-05-25:
It was good on the test server.
It is good now on the prod server since Monday morning.

mregen, 2022-06-01:
please check with 1.4.368.58 and reopen if there are still issues!

closed 2022-06-01 by mregen