OPCFoundation/UA-.NETStandard #1786: OPCUA client does not check the session id returned in the createSessionResponse

GregoireG-C, 2022-04-27:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [X] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

OPCUA client does not check the session id returned in the createSessionResponse.

### Expected Behavior

The client should not create the session if the sessionId is not consistent.

### Steps To Reproduce

1 Compile the referenence standard OPCUA client
2 Compile the reference standard OPCUA server
3 Run the server
4 Use the CTT client to test the client and run the test Err-003 from Session Client Base conformance unit
5 Run the client and create two sessions

=> You should observe that the client create both sessions with different session id.

### Environment

```markdown
- OS:Windows
- Environment:Visual Studio 2019 16.7.2
- Runtime:NET 4.8
- Nuget Version:
- Component:Opc.Ua.Client
- Server:Reference Server
- Client: Reference client
```

mregen, 2022-06-11:
@GregoireG-C , do you have a fix to recommend?

GregoireG-C, 2022-06-13:
No, sorry, I didn't take a look at the code.

GregoireG-C, 2022-12-06:
Well I added the following code after the CreateSession in Session::Open:
```
if (sessionId.IsNullNodeId)
                throw ServiceResultException.Create(StatusCodes.BadSessionIdInvalid, "Server has returned a null sessionId.");
```
According to the standard (8.2.4 Part 3):

> A Node in the AddressSpace shall not have a null as its NodeId

Which is the case for SessionId

Do you aggree with us?

Regards,

closed 2022-12-27 by GregoireG-C