OPCFoundation/UA-.NETStandard #1790: GDS API  to query Registered Application's  certificate provisioned or not state ? and validity details

charlesaugustineabb, 2022-04-28:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

This query is regarding spec 12 GDS.
1)Is there any GDS API provides information like registered UA application pulled or pushed the signed certificate?
2) Is there any GDS API  provides information like specific Registered application's certificate(if certificate is already provisioned) validity details like  valid from" and "valid to" dates?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

charlesaugustineabb, 2022-05-25:
Kindly help me with Answers, if possible  . This Question is open from long time.

charlesaugustineabb, 2022-05-31:
Spec 12 version 1.05.02 , when will be released?  is there a possibility to get sample SDK & samples of new version?

mregen, 2022-06-11:
@charlesaugustineabb the latest Part 12 update is still being discussed. Having the GDS sample updated is a community effort, we need some volunteers who would like to help...

charlesaugustineabb, 2022-06-13:
I can be a volunteer for the above GDS development activity.

mregen, 2023-07-27:
Hi @charlesaugustineabb, the 1.05.03 spec is now a release candidate. I don't believe there is now support for the API you asked above. For missing functionality I'd suggest to open Mantis issues (https://mantis.opcfoundation.org/) in the OPCF bug database, so it can be picked up by the WG.
For updating the open source code to the latest spec would be greatly appreciated. Please let @EthanChangAED know which feature you would like to pickup and work on so we can coordinate the efforts. Thanks, Martin

mregen, 2024-02-27:
Hi @charlesaugustineabb, please check the Februar release with the recent GDS and spec updates if this is still an issue.

EthanChangAED, 2024-03-29:
closing this one as fixed. feel free to reactive if there is still any functionality missing.

closed 2024-03-29 by EthanChangAED