OPCFoundation/UA-.NETStandard #1795: Problem with Subscription PublishStatusChanged events without changing item value

RoseauBruyant, 2022-05-03:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Subscription PublishStatusChanged events : There is no change in value, but events occur (at the rate of keep alive)

### Expected Behavior

If there is no change in value, there is no event

### Steps To Reproduce

I observed a problem in my applications. I have tried different settings without success.
I present the problem here using available sources.

1. With sources "UA-.NETStandard-Samples-master", opening solution "UA Sample Applications.sln".
Selecting the "UA Sample Server" project on startup and launching.

2. Copy of project "UA-.NETStandard-Samples-master", launch of solution "UA Sample Applications.sln".
Selecting the "UA Sample Client" project at startup and launching.

3. In the client, enter the URL of the server "opc.tcp://desktop-XXXX:51210/UA/SampleServer" and select the first choice "SignAndEncrypt" "Basic256Sha256" ... and connect "Anonymous"

4. in "MySession 1",Select the variable "Objects/Data/Static/Scalar/Byte Value", right click "Subscribe>New".

5. problem: In project "UA Sample Controls/Subscriptions/SubscriptionsDlg.cs"
in the function, void Subscription_PublishStatusChanged(object subscription, EventArgs e)
With a spy, we observe that the event occurs without changing the item.

is there a bug? A specific parameter to fill in ? A mistake from me ?

mregen, 2022-06-11:
@RoseauBruyant I think thats a bug... but needs investigation.

RoseauBruyant, 2022-06-28:
thank you for your reply