OPCFoundation/UA-.NETStandard #1800: Server always returns endpoints with hostname

mregen, 2022-05-06:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When doing a discovery on your Server with the IP endpoint it returns endpoints with a DNS name. It would be nice if the server would return that endpoint with the IP as requested.

### Expected Behavior

Endpoints contain ip address

### Environment

```markdown
- Nuget Version: any
- Component: GetEndpoints service
- Server:
- Client:
```

### Anything else?

[image]

GregoireG-C, 2022-06-28:
This behaviour had been present in V1.02 and you changed it in V1.03. 

I don't hunderstand you point tho. Why it would be nice for the client to get the exact endpoint that it requested ?
The norm says that:

> The implementation of this Service shall use this information to return responses that are accessible to the Client via the provided URL.

So, as the server gives an accessible DSN name with an applicative certificate which show the same DNS name. It should work.

Ame I missing something ?

mregen, 2023-04-16:
Hi @GregoireG-C, there are cases when the name resolution is not working. Then the returned endpoint Url should also use the IP address that was passed in the GetEndpoints call. Or if the server can only be reached using the FQDN, then the endpoints should use that FQDN for the endpoints. However, only if the server specified these Urls for the alternate hostnames.

PR #2123 (2023-04-18): Improvements for docker hosting and fix for GetEndpoints 
## Proposed changes

- Add environment variables as parameter to support docker hosting where command line parameter are not available
- bump some Nuget packages
- only warn on cert validation error on startup, e.g. when the CRL becomes outdated allow to start and print warning.
- Let GetEndpoints return endpoints which match the DNS name used. DNS names must be specified as alternate hostnames in the configuration.

## Related Issues

- Fixes #1800 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in th [... cut]
closed 2023-04-18 by mregen