OPCFoundation/UA-.NETStandard #1808: How can i add ModelingRules

GregoireG-C, 2022-05-11:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I want to add the standards modelling rules into the ModellingRule folder but I don't see how to do so.

Can you help me on this ?

### Expected Behavior

[image]

### Environment

```markdown
- OS:Windows 10
- Environment:Visual Studio 2019
- Runtime:NET 4.6.2
- Nuget Version:1.4.367.41
- Component:
- Server:Reference server
- Client: UA Expert
```

GregoireG-C, 2022-05-12:
Using a new class which implement StandardServer class :

I added the following code :
```
protected void CreateModellingRules(Opc.Ua.Server.IServerInternal server)
{
Opc.Ua.FolderState state = ModellingRuleFactory(Opc.Ua.BrowseNames.ModellingRule_ExposesItsArray, Opc.Ua.ObjectIds.ModellingRule_ExposesItsArray, 
Opc.Ua.ObjectIds.ModellingRule_ExposesItsArray.NamespaceIndex);
server.ServerObject.ServerCapabilities.ModellingRules.AddChild(state);
[...]
 }
```

```
protected Opc.Ua.FolderState ModellingRuleFactory(string sBrowseName, Opc.Ua.NodeId nNodeId, ushort uNameSpace)
        {
            Opc.Ua.FolderState state = new Opc.Ua.FolderState(null);

            state.SymbolicName = sBrowseName;
            state.ReferenceTypeId = Opc.Ua.ReferenceTypes.HasComponent;
            state.TypeDefinitionId = Opc.Ua.ObjectTypeIds.ModellingRuleType;
            state.NodeId = nNodeId;
            state.BrowseName = new Opc.Ua.QualifiedName(sBrowseName, uNameSpace);
            state.DisplayName = state.BrowseName.Name;
            state.WriteMask = Opc.Ua.AttributeWriteMask.None;
            state.UserWriteMask = Opc.Ua.AttributeWriteMask.None;
            state.EventNotifier = Opc.Ua.EventNotifiers.None;

            return state;
        }
```

[image]

Do you aggree with its compliance ?

GrG

closed 2022-12-06 by GregoireG-C

ahaas207, 2024-02-08:
Hi @GregoireG-C  I know this has been closed for a while, but I was hoping you could let me know what resolution you landed on for adding this in. I am on a team that is working towards test lab certification and as you have noted, this is required for compliance. 

Following a similar process to the code snippet above, I am able to browse the 5 modelling rules in the Server Capabilities - Modelling Rules folder, however I am having trouble with the PropertyState NamingRules. I am able to add a NamingRule as a child to each ModellingRule, but they are entirely missing when browsing. Could you share how you approached the NamingRules aspect? Also, was the above route sufficient for compliance? 

Thank you for your time,
Aimee