OPCFoundation/UA-.NETStandard #1823: MonitoredItemCount differs in this stack and UAExpert<title>

TeaTime762, 2022-05-18:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

I have built a blazor application on top of your opc-ua client stack. 
When switching pages the monitored items from the last page are removed and the monitored items of the current page are added into the subscription. 
When I have a look in the MonitoredItemCount via the class hierarchy Session -> Subscription (select matching subscription) -> MonitoredItems -> MonitoredItemCount I get the expected MonitoredItemCount
[image]

But when looking in UA Expert it always says to me that there are more MonitoredItems than I expected.

[image]

To which variable targets the MonitoredItemsCount in the class hierarchy?
Is this expected behaviour or do I miss something?

### Expected Behavior

I would expect that this client stack displays the same amount of MonitoredItems than the UA Expert client.

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022 17.1.4
- Runtime:.NET 6
- Nuget Version: 1.4.367.75
- Component:
- Server: Beckhoff TF6100 | TwinCAT 3 OPC UA
- Client: This stack / UA Expert 1.6.2
```

mrsuciu, 2022-06-07:
@TeaTime762 
The MonitoredItemCount's displayed in UAExpert is a variable maintained by the server, while the Session -> Subscription (select matching subscription) -> MonitoredItemCount is a Property of the Client.Subscription (client side) which keeps track of the number of monitored items (per selected subscription) on the client side with no direct relation to the one displayed by UAExpert. Since you see a different value on the server through UAExpert most probably the source of the discrepancy is on the server implementation side.

TeaTime762, 2022-06-07:
Thanks for your explanation!

closed 2022-06-07 by TeaTime762