OPCFoundation/UA-.NETStandard #1829: How can i read variables?

WanderWashington, 2022-05-28:
Helllo!

I'm trying to read variables from a specific object using the Console Reference Client, Do you have some example doing it ? Not using the server node, but a specific folder inside the objects.

In my case, inside the objects, i have some folders and i have to get the values inside the variables
[image]

mrsuciu, 2022-06-07:
@WanderWashington 
Under the UAClient there is the ReadNodes() method which presents two ways to read the value attribute for nodes that are Identified by their NodeId.
In your case you would probably need to Browse ( Browse() method example) from a well known NodeId or use TranslateBrowsePathsToNodeIds service to find out the NodeId's that identify the targeted nodes.

WanderWashington, 2022-06-22:
@mrsuciu  Thanks a lot.  It helped me so much 👍 . I used the Browse method to discover the NodeIds and the folders and the method ReadValue, for read each NodeId...

closed 2022-06-22 by WanderWashington