OPCFoundation/UA-.NETStandard #1832: BinaryDecoder creates different result with same input as UAExpert-Stack

JonasReichhardt, 2022-06-01:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I currently have a problem with our self-written client calling a method an a propiertary OPC-Server.
Calling the method with the UAExpert it works without a problem.

After tracking the sent data via WireShark I found that the ByteString which is sent is differs between my client and the UAExpert.

UA Expert seems to stuff non-zero bytes between decoded items, where this stack uses zero bytes. (Delimiter??)

This is our own implementation
[image]
and this the UAExpert
[image]

It can be seen in the pictures that structure-wise the packets are the same
We tried a update of the stack from 1.4.362.38 to 1.4.368.58, due to the fact that we moved the project to .Net Core 6.0 and thought that this might be the reason.

I'm happy to provide as much information as I can. 
Just making sure that there is no problem with our implementation or the stack before contacting the vendor of the OPC-UA Server.

### Expected Behavior

The same input should result in the same sent ByteString

### Environment

```markdown
- OS:Windows 10
- Environment: VS2022
- Runtime:.Net Core 6.0
- Nuget Version: 1.4.368.58
- Component: Opc.Ua.Core
- Server: proprietary
- Client: Self-written / UA Expert
```

mregen, 2022-06-11:
@JonasReichhardt could ou provide code snippets how you build and call the service?

JonasReichhardt, 2022-06-14:
Note: I forgot to mention that the method in question has an array of custom objects as input parameters. 

This specific call works on server from other vendors and generally the method calling functionality has no problem with other methods and input parameters.

We made a wrapper for most of the OPC Stacks functionality.
_Session creation_
```
var session = new Session(
          GetTcpChannel(applicationConfiguration, configuredEndpoint),
          applicationConfiguration, configuredEndpoint,
          null)
        { ReturnDiagnostics = DiagnosticsMasks.All };

session.Open($"{driverName ?? ""}",
        (uint)session.SessionTimeout,
        endpointOptions.GetUserIdentity(), null, false);
```

GetTcpChannel creates an SessionChannel object

_Function calling_
```
public IList<object> Call(NodeId objectId, NodeId methodId, params object?[] args)
{
   return Session.Call(objectId, methodId, args);
}
```

If you need more information just let me know

benjamin1988, 2022-07-25:
Is their any further information? I am fighting with the same issue.

JonasReichhardt, 2022-12-12:
The problem was on the server side. Therefore I will close the issue

closed 2022-12-12 by JonasReichhardt

mregen, 2022-12-12:
> The problem was on the server side. Therefore I will close the issue

could you share the root cause on server side, would be good to know for others...

JonasReichhardt, 2022-12-12:
I contacted the vendor and they had faulty decoding code with tried to decode a uint as a string. Therefore the called Method threw an exception.

So it seems that in the end the difference in sent data does not matter in real world applications